Ultimaker/Cura #4232: [3.4.1] Machines profiles, variants, materials

Tofoillote, 2018-08-13:
V3.4.1 Win 7 64 Bit - Creality Ender 3 (based on CR10 tweaked)

Please do publish document structure about creating Variants.

I wanna taylor made my cura version in accordance to material I own and use with diverse nozzle ø.

So wanna have only 1 printer and a few material, not a whole library.
This is in order to increase experience return capability.

As is, "basic" settings are too basics then intermediate nor expert are tinker tanks.

So I wanna couple materials and printer as easy as possible. without having to re-check the whole setting list or nozzle ø.

[creality_Ender3.def.json.zip](https://github.com/Ultimaker/Cura/files/2282523/creality_Ender3.def.json.zip)

ChrisTerBeke, 2018-08-13:
Some documentation on this is needed indeed, we're actually working on that.

The 'advanced' settings can already be tuned and have some presets based on experience level.

nallath, 2020-01-30:
https://github.com/Ultimaker/Cura/wiki/Profiles-&-Settings

closed 2020-01-30 by nallath