Ultimaker/Cura #4235: Change quality settings mid-model (feature request)

adiastra, 2018-08-13:
**Application Version**
3.4.1

**Platform**
Windows 10
**Printer**

Monoprice Mini Delta
**Steps to Reproduce**
feature request, please see additional information 

**Actual Results**

**Expected results**

**Additional Information**
Would like to be able to print different parts of model at different quality. A use case for this would be printing the base of a miniature in coarse quality and the detail of the actual model in fine quality. Being able to select quality based on  vertical regions would be very useful.

ChrisTerBeke, 2018-08-13:
This has been requested often and because of issues in both software architecture and print quality, we're most likely not going to do this.

adiastra, 2018-08-13:
So much for building your product with your community.

ChrisTerBeke, 2018-08-13:
Anyone in the community is free to help us build this if they see a way to properly do this. We have to prioritise our time to keeping Cura stable and working on features for Ultimaker.

Ghostkeeper, 2018-10-12:
This is not a duplicate as far as I know. If it is, please indicate the ticket that this is a duplicate of.

It is considered pretty difficult though, because we have no system to set settings per layer, no interface for it, and some settings that cause layers to have influence over other layers (such as Infill Layer Thickness).

lexx9999, 2023-07-12:
> This has been requested often and because of issues in both software architecture and print quality, we're most likely not going to do this.

There are some many settings that affect print quality to negative that I would decline that argument.

In general I think such a feature would be useful.

What about an additional "Mesh Type", "Modify settings by included layers" (With options "Included layers only" / "From lowest layer to all above") ? Or just "Modify settings from included lowest layer" 

Would by quite clear, still only minimal UI change.
A warning if multiple objects at the same layer of such a type would be possible without UI change.

GregValiant, 2026-09-06:
I'm cleaning up some of these old reports.  I'll close this.  Currently 5.13.0 is the latest Cura version.

In the Cura MarketPlace is a plugin called "HellaFusion" by the Cura contributor _HellaHolic_.  It pulls "saved" profiles in and changes the profile at the heights you indicate in the setup.  You can also insert pauses in case you want to change nozzles.
Make no mistake - it is "Advanced Cura" and there is a learning curve.  It does allow you to make massive changes mid-print.

closed 2026-09-06 by GregValiant