Ultimaker/Cura #4240: One at a Time and Raft

FynDestiny, 2018-08-15:
**Application Version**

**Platform**

**Printer**

**Steps to Reproduce**

**Actual Results**

**Expected results**

**Additional Information**

BagelOrb, 2018-08-16:
This might very well be a rendering issue. The raft layer numbers are negative which might cause problems in the frontend.

FynDestiny, 2018-08-16:
Surely when the gcode is generated, the engine would take that into consideration and adjust the layers accordingly

BagelOrb, 2018-08-16:
Actually I think I've seen this rendering issue before.

Check the gcode in another viewer like repetierHost or in a web browser visit tinyg chillipeppr.

FynDestiny, 2018-08-20:
I'll look into it Bagel thanks

I must admit, if I knew how to do it, I'd write my own version of Cura, but my programming skills, or rather my knowledge in programming is only average. If I could have a copy of all the code of Cura, I'd love to be able to learn more. Don't know if that's possible or not though ?

BagelOrb, 2018-08-20:
Haha. All the code is online right here. Click on the tab <>code at the top left.

Cura is split into several parts. The main parts are Cura (this repo), Uranium and CuraEngine. The former two are in python and the latter one is in c++.

However, the amount of code is vast and a novice programmer will likely soon get lost and drown in all the code. Maybe you can start writing a plugin to learn a bit more about the code. There are loads of Cura plugins out there on github to learn from.

FynDestiny, 2018-08-20:
I'm not a "complete" novice, I've used Standard & Visual Basic, Java, Assembler (that was fun), some C & C++, a little Python, website languages and a few others I can't remember.

So it will be interesting to read through it, and attempt to rewrite it in, say, Java, as I have been recently studying it (that's how I learn a language better, makes me look stuff up).

BagelOrb, 2018-08-20:
You could write your own slicer based on the code here, of course, but I'm
not sure whether that would be fruitful.

Maybe it would be more satisfying to you to contribute to the Cura project
rather than starting your own. Knowing that code you have written ends up
being used all over the world is actually quite satisfying.

That way you don't start an endless project and you get help from us when
you made a mistake. You'll learn quicker and you'll get more satisfaction
out of it.

It's just an idea..

Ghostkeeper, 2018-10-24:
This issue was actually not a rendering issue, but an actual g-code that was extruding a line towards 0,0. This was fixed in version 3.5.1.

closed 2018-10-24 by Ghostkeeper