Ultimaker/Cura #4242: Double Z Hop on Tool Change

vjapolitzer, 2018-08-16:
**Application Version**
3.4.1

**Platform**
Win10

**Printer**
Custom, Multi-Extruder, RepRap Gcode flavor

**Steps to Reproduce**
Enable Z Hop when retracted, as well as Skirt or Brim. Use a multi color model where at least the first two layers use only one color/extruder (I used the Low-Poly Eevee by FLOWALISTIK). Ensure the selected Build Plate Adhesion Extruder is the same as the extruder used for the first two layers of the model. For the Eevee, I used Extruder one for eevee...dual1.stl and Build Plate Adhesion, and Extruder two for eevee...dual2.stl. Then slice the model and export the Gcode.

**Actual Results**
In the transition from the 0th to the 1st layer, when the tool is changed from the outer skirt/brim tool(s) back to the tool for the model, an additional Z Hop line is present, so a double Z Hop occurs. An example is in the _Additional Information_ section.

**Expected results**
Only one Z Hop should be present.

**Additional Information**
Project File:
[CFDMP_eevee_lowpoly_flowalistik_dual.curaproject.curaproject.3mf.zip](https://github.com/Ultimaker/Cura/files/2292500/CFDMP_eevee_lowpoly_flowalistik_dual.curaproject.curaproject.3mf.zip)
Example Gcode:
;TIME_ELAPSED:67.119864
;LAYER:1
M140 S60
M106 S255
G0 X0.00 Y0.00 Z0.7
G1 F2700 E12.83334
G92 E0
T0
G92 E0
M109 S200
**G0 F24000 X0.00 Y0.00 Z0.7** _First expected Z Hop (2 * 0.3mm layers + 0.1mm hop)_
**G0 X0.00 Y0.00 Z0.8** _Unexpected second Z Hop_
G1 F2700 E-3.2
G0 F24000 X0.00 Y10 Z0.7
G1 F30000 Z0.7
G0 F24000 X-5.613 Y-9.448
;TYPE:WALL-INNER
G1 F30000 Z0.6
G1 F2700 E0
G1 F1950 X-4.688 Y-8.845 E0.05509

Ghostkeeper, 2018-08-16:
When slicing this I get an assertion failure in CuraEngine.

`2018-08-16 08:34:23,582 - DEBUG - [Thread-15] UM.Backend.Backend._backendLog [94]: [Backend] CuraEngine: /home/ruben/Projects/CuraEngine/src/LayerPlanBuffer.cpp:93: void cura::LayerPlanBuffer::addConnectingTravelMove(cura::LayerPlan*, const cura::LayerPlan*): Assertion `newest_layer->extruder_plans.front().extruder == prev_layer->extruder_plans.back().extruder' failed.`

We don't show these assertion failures unless you're in debug mode, but something like this would cause weird results.

BagelOrb, 2018-08-16:
I can't get my debugger to run :cry: 

I can't continue working on this issue, so I'm leaving my findings for future developers who want to pick up the issue.

The problem is definitely there:
```
;LAYER:1
M140 S60
G1 F2700 E12.83336
G92 E0
T0
G92 E0
M106 S255
G0 F24000 X0.00 Y0.00 Z0.7
G0 X0.00 Y0.00 Z0.8
G1 F2700 E-3.2
G0 F24000 X0.00 Y10 Z0.7
G1 F30000 Z0.7
G0 F24000 X6.286 Y-12.249
;TYPE:WALL-INNER
```

It seems to be related to the fact that the machine doesn't need any action to switch extruder, which caused an empty extruder plan, which causes assertion failures in `LayerPlanBuffer::addConnectingTravelMove`.

The extruder plan is removed in `LayerPlan::setExtruder` if there are no travel/extrusion moves in the plan, but this causes problems with assertions.

Ghostkeeper, 2018-08-27:
Two of our engineers have worked on this but couldn't fix the bug. We've decided to leave it in for now but provide a workaround: In the Extruder Start g-code of both extruders in the Machine Settings menu, put down this g-code:

    M105

The disadvantage it that it'll switch to the other extruder and back unnecessarily at some points. If this is a no-op on your printer then you're fine but you will have some unnecessary heating and on some printers an unnecessary move.

closed 2023-06-24 by github-actions[bot]