Ultimaker/Cura #4246: [3.4.1] - Dual Nozzle 2 in 1 out (Cyclops) settings

MachuRoberts, 2018-08-16:
**Application Version**
3.4.1

**Platform**
Windows 10

**Printer**
Custom Printer

**Steps to Reproduce**
- Setup printer with dual extruder, using a 2-in-1 out system (Such as E3D Cyclops)
- expose ALL temperature settings. this prevents Cura from thinking there are two nozzles and keeping one at a lower temp
- set model and print

**Actual Results**
- when the printer ends with one extruder, and is preparing for the second extruder to prime in the tower, the M109 command pauses the printer and causes the hot end to heat up (the 1 or 2 degrees that it may have changed during print)
- this cause the printer to pause in place for 1-3 seconds as the M109 command pauses the printer until the temperature is reached

**Expected results**
- a setting which would tell Cura not to place the M109 command in the Gcode such as "2-in-1-out Nozzle"
- when this setting is activated there could be multiple other settings that would significantly help fine tune the result when using such a hot end.
- removing this M109 command would allow the printer to continue as it would have normally, and have the hotend cycle to heat back up without pause

fieldOfView, 2018-08-16:
This can also be done using a postprocessing script.

MachuRoberts, 2018-08-16:
yes, it sure could...but as a feature request and borderline bug, it would
be great to have the option for Cura to know its a single nozzle and
therefore not add complications to the gcode.

ChrisTerBeke, 2018-08-16:
This has been asked before and the team at Ultimaker is not going to spend time on this as it's not relevant for Ultimaker hardware.

closed 2018-08-16 by ChrisTerBeke