Ultimaker/Cura #4247: Cure slicing missing layers

EliasDebaveye, 2018-08-16:
**Application Version**
3.4.1

**Platform**
Windows 10

**Printer**
Ender 3
[image]

smartavionics, 2018-08-16:
Well, it could be the model or all sorts of things. Tell  you what, if you attach the project file, someone could possibly find out what's going on.

EliasDebaveye, 2018-08-16:
[Picachu_Remix_Low_poly.zip](https://github.com/Ultimaker/Cura/files/2295407/Picachu_Remix_Low_poly.zip)

smartavionics, 2018-08-17:
Thanks for the model file. The model has flaws, that's the problem. This is what Freecad says about it:

[image]

Ghostkeeper, 2018-10-16:
Yes, it seems that the nose was pasted on afterwards using loose polygons rather than a closed volume. This causes it to become ambiguous which parts of the model are inside the model and which parts are outside. In your case, it decides that most of the head is outside the polygon there.

You can see this in X-ray view pretty easily.
[image]

closed 2018-10-16 by Ghostkeeper