Ultimaker/Cura #4248: white menu

Burning-man, 2018-08-16:
Hi all, I have a probleme with Cura 3.4.1

I already see some poeple have the same probleme as me but they solved it without telling how they do that...
When I launch cura all is good but after I check some of the toolbar boutons the box became white and all the falling menu become white...

Some says it's the dpi of the application but I don't know how to change that, I'll already made all the updates for graphic drivers, try launching cura with changing the ppp with windows, changing parameters with the nvidia sofware, uninstall completly (all the files found in conenction with cura)

Please can you help me?

**Application Version**
3.4.1

**Platform**
Windows 10 64bit, I5 6600K, GTX1080, 16Go ram, screen display on 1080p 1920x1080

**Printer**
Tevo Black Widow

**Steps to Reproduce**
just clic on the tool bar on top and slide from left to right for displaying falling menu

**Actual Results**
menu became white, right clic too and all falling menu too

**Expected results**
nothing like these should happen

**Additional Information**
I have the problem since the launch of the new cura in 2017

Burning-man, 2018-08-19:
I have a new info about my probleme, in the log I have this at the end

2018-08-19 19:48:31,450 - DEBUG - [MainThread] cura.Machines.MachineErrorChecker._rescheduleCheck [129]: New error check scheduled.
2018-08-19 19:48:31,750 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [301]: file:///D:/Program Files/Ultimaker Cura 3.4/resources/qml/Settings/SettingCategory.qml: Object destroyed during incubation
2018-08-19 19:48:32,170 - INFO - [MainThread] cura.Machines.MachineErrorChecker._setResult [181]: Error check finished, result = False, time = 0.7s
2018-08-19 19:48:41,515 - DEBUG - [MainThread] cura.AutoSave._onTimeout [49]: Autosaving preferences, instances and profiles
2018-08-19 20:12:23,666 - DEBUG - [MainThread] UM.Qt.QtApplication.windowClosed [379]: Shutting down cura
2018-08-19 20:12:23,684 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [276]: Attempting to kill the engine process
2018-08-19 20:12:23,684 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [282]: Killing engine process
2018-08-19 20:12:23,688 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [285]: Engine process is killed. Received return code 1

Peharps someone can help me with this?

fieldOfView, 2018-08-20:
Nothing there is out of the ordinary.

Burning-man, 2018-08-20:
the "warning" too?

Burning-man, 2018-08-20:
here it is, https://pastebin.com/Rzyky1HZ

I just upgrade my GPU's driver last week, but before it don't work either...

fieldOfView, 2018-08-20:
Nothing saucy in there either.

Have you ever used MeshMixer on the same system? It uses the same GUI framework, but a newer version. It would be interesting to know if that exhibits the same issue.

Burning-man, 2018-08-20:
I just try it, no any problems seen on screen... and all seem to be fonctionnal, log is cleen when I click on the tab "show log"

Ghostkeeper, 2018-10-24:
We're not able to reproduce this. It seems to be specific to a certain graphics driver or something.

Burning-man, 2018-12-05:
Sorry for long time without answer...

Yes, ever in 3.6 problem still here, so I saddly continue to use old 15.04 cura without all the possibility of the new one and it's really frustrating :(

Burning-man, 2018-12-07:
here it is a log from cura 3.6.0

https://pastebin.com/SUub3rGC

if it can help

onitake, 2020-02-08:
Isn't this the same as #5296 ?
Do you still see it in Cura 4.x?

Ghostkeeper, 2020-02-12:
Yeah, looks like it. We learned some new things about this issue more recently, which are listed in #5296. I'll close this as a duplicate, so please continue discussing it there.

closed 2020-02-12 by Ghostkeeper