Ultimaker/Cura #4249: 3.4.1 Filament Change post processing has backwards g-code for extrusion/retraction

boomercat, 2018-08-17:
**3.4.1**

**Ubuntu 16.04**

**MP Mini Select**

**Steps to Reproduce**

**Post Processing > Plugins > Filament Change (I've tried this for changing at different levels and it is always the same result.)**

**When the printer gets to where it should pause and retract the filament it actually pauses then extrudes filament. After the filament change, when I hit continue, it then retracts the filament. This causes me to have to manually feed the filament or my print will create a big hole. Also, the Z axis setting that it returns to is much too high because subsequent layers after the filament change have no "smush" and simply sit on top of the previous layers (with a noticable gap). This causes a massive split to form in the print.**

**I could take a video or something if somebody needed it.**

putridp, 2018-08-23:
I've also noticed this problem of the retract/extrusions being reversed - major clicking of the extruder then a retration as it starts the next layer.

Also, is it correct that the action takes place at the beginning of the specified payer? In contrast, the 'Pause at Height' post processing script, when specifying the layer, carries out the pause at the end of the layer. It would be good for the behaviours to be consistent, or at least specified in the settings

Ghostkeeper, 2018-10-25:
Cura's post-processing script doesn't actually determine the order in which these things happen. That's in your firmware. All we output is something like this:

    M600 E-30 L-300

That indicates that there should be a short retraction of 30mm before the switch, and a long retraction of 300mm to remove the material from the printer. Note that we do make the distance negative. Apparently Cura interprets the E and L parameters as a delta position for the extruder wheel, so a negative value would retract it.

It looks like Marlin, by default, doesn't care about what direction the retraction is going (notice the `ABS()` call):

https://github.com/MarlinFirmware/Marlin/blob/5cf6c44f47737bdf93df6ae7c835fe2d54d0675d/Marlin/Marlin_main.cpp#L10993

I don't know what heritage of firmware MP Mini is using or Putridp's printer, so maybe it's wrong for that firmware. Also the printers at Cura's development team are all Ultimaker printers that don't even support this command and completely ignore it, so we can't test for it.

I'll believe you that the distance should not be inverted. That also seems more logical to me. I'll remove the negative signs.

Ghostkeeper, 2018-10-25:
The Z position definitely is a bug in your printer's firmware. Cura doesn't specify anything with regards to the Z position.

commit c7b146c9 (2018-10-25): Don't make retract distances negative

People are reporting that the retraction is reversed. We can't test it since we have no firmware that supports the command at all. So we'll just not make the retract distance negative. That seems more logical anyway if you read the documentation of M600 to the letter.

Fixes #4249.
closed 2018-10-25 by Ghostkeeper