Ultimaker/Cura #4252: STL file loads and slices but it wont save on SD card

fjimmy0, 2018-08-17:
The program works just fine then when i try to save it to the SD card, an error occurs. Now i cant print anything.
[image]

[image]
How do i fix this? Should i reinstall the program?

LipuFei, 2018-10-10:
Hi @fjimmy0 Could you upload file? You can find it in
```
C:\Users\<your-user-name>\AppData\Roaming\cura\<cura-version>\cura.log
```

And, if you try to save it to a local hard drive, does it work?

LipuFei, 2018-10-10:
Could be related to https://github.com/Ultimaker/Cura/issues/4495

LipuFei, 2018-10-10:
Hi @fjimmy0 We have a candidate build for Cura 3.5.1, could you try it? We have a fix for #4495 in it and I wonder if it works in your case. Thanks!
Link: https://we.tl/t-0gqXweB9KG

LipuFei, 2018-10-11:
I will close this issue because it's probably the same as #4495. That has been fixed for the next release 3.5.1. You can reopen it in case it's not the same issue.

closed 2018-10-11 by LipuFei