Ultimaker/Cura #4253: [3.4.1] Anet A8 smashed through 3D print partway through

simorph, 2018-08-18:
Reporting this just in case someone else experiences this - been using Cura for nearly a year now and never had such a wild behaviour/catastrophic fail.  Sending you the info so you can perhaps improve error checking to avoid the printer seemingly driving the print head through a model.

**Application Version**
3.4.1

**Platform**
Windows

**Printer**
Anet A8 running SkyNet firmware

**Steps to Reproduce**
Loaded up STL file (an R2D2 dome part from Thingiverse) and reoriented to print with the base on the build plate (sadly the STLs aren't supplied with a simple orientation). Exported as usual and set off my print.  Everything looked fine in the print preview within Cura.

**Actual Results**
Just over halfway through the about 15-hour print the printer decided to ram the print head through the model.  Luckily I managed to hit stop on the power before much damage could occur.

I figured that it was probably something gone awry in the gcode.  So I pulled the gcode into an online viewer (http://gcode.ws/) and immediately saw a giant space in the middle of the image that seemed to confirm my suspicion that something had gone awry - the gap in the model in the view is at the point that the printer went crazy.

**Expected results**
Seamless print as usual, no print head demolishing the model partway through!

I have since reimported the part, reoriented it and sliced it with Cura and viewed the new gcode - no gaps in the model so I'm suspecting that it's Cura and not a STL issue, hence why I'm posting this for you.

**Additional Information**
Attached image from the gcode analyser - clearly a big space in the model - the print failed at the moment that it reached that horizontal gap in the model.
[image]

Attached STL file and the GCODE generated by Cura that failed
[STL and GCODE.zip](https://github.com/Ultimaker/Cura/files/2299525/STL.and.GCODE.zip)

This contains my second slice of the file which seems to show a successful slice
[NEW GCODE.zip](https://github.com/Ultimaker/Cura/files/2299531/NEW.GCODE.zip)

Hope this helps!

SharkSteps, 2018-08-24:
I have a similar problem, but only if i export STL from Fusion 360 in High, if i export in medium i have no problem with cura, but now i have a job in my printer, and it stops, i need to continue, because is a 10h job, i don't know how can i start the print from that layer.

Ghostkeeper, 2018-10-25:
Whenever you have such problems, be sure to look at the STL in X-ray view in Cura to see if there are no red areas. Red areas means that there is extra geometry on the inside of your model or that the model is not water tight. That typically creates a result like this.

In Simorph's case though it appears to be something else, since there are moves towards 0,0,0. Cura doesn't normally create moves to a different Z position unless changing layers or Z-hopping, so this is very weird to me. You didn't happen to have a post-processing plug-in active?

I'm not reproducing the issue here. Your R2D2 model sliced fine for me in the current Cura 3.6 release candidate. Maybe you can try again using 3.5?

nallath, 2019-03-01:
Closing due to inactivity.

closed 2019-03-01 by nallath