Ultimaker/Cura #4254: Insufficient clearance for prime blob with 0.8mm print core

mqudsi, 2018-08-18:
**Application Version**
4.3.1

**Platform**
Windows 10 x64

**Printer**
Ultimaker 3 Extended

**Steps to Reproduce**
Print with 0.8mm print core and prime blob enabled

**Actual Results**
There is insufficient clearance for the prime blob, the print head ends up swimming in the ejected blob.

**Expected results**
The print clearance should take into account the print core diameter, flow rate multiplier, and possibly even material density/plasticity.

fieldOfView, 2018-08-19:
You mean 3.4.1, right?

mqudsi, 2018-08-19:
I do indeed.

Sembiance, 2018-08-21:
I have this same problem but with the 0.4mm BB core.

Is there any way to modify how high above the print bed the blob is extruded from? I'd like to raise it higher in general to avoid the blob touching the exterior of the hot end.

BagelOrb, 2018-08-23:
@sembiance: please create a new ticket for your request.

BagelOrb, 2018-08-23:
@mqadsi you are talking about clearance in the XY direction, right?

mqudsi, 2018-08-23:
No, in the z.

diegopradogesto, 2019-09-11:
Long time waiting for a reply. Is this still causing you problems, @mqudsi ? 

Feel free to reopen if necessary.

closed 2019-09-11 by diegopradogesto