Ultimaker/Cura #4255: Concentric 3D support generates incorrect results

mqudsi, 2018-08-18:
**Application Version**
3.4.1

**Platform**
Windows 10 x64

**Printer**
N/A

**Steps to Reproduce**
Require support midway through an object (i.e. not from the start) and specify concentric 3D as the support pattern.

**Actual Results**
There are a few of layers of support touching the build plate where the support should begin, then nothing before the support resumes in mid-air.

**Expected results**
A proper support tower should have been generated.

**Additional Information**

Concentric 3D support:
[image]

Any other support:
[image]

Sample project:
[UM3E_Masonry Sign v7_Masonry Sign v7_Body70_Masonry Sign v7.zip](https://github.com/Ultimaker/Cura/files/2299910/UM3E_Masonry.Sign.v7_Masonry.Sign.v7_Body70_Masonry.Sign.v7.zip)

BagelOrb, 2018-08-23:
Why do you want concentric support?

mqudsi, 2018-08-23:
Who says I do?

mqudsi, 2018-08-23:
I was only exploring when I happened on this bug. Thanks for taking the time to look at it in all cases.

BagelOrb, 2018-08-23:
Thanks for reporting.

Ghostkeeper, 2018-10-25:
We've removed Concentric 3D as option there so this is no longer a concern now.

closed 2018-10-25 by Ghostkeeper