Ultimaker/Cura #4256: Add support placement option that does not require touching build plate

mqudsi, 2018-08-18:
(As of Cura 3.4.1)

It would be great to have an option to generate support in a manner that does not require it to touch the build plate at all. Currently the options are "touching the build plate" and "everywhere", both of which generate extraneous support that adds a component to the print time. 

Here's an example of what is currently needlessly generated for a particular configuration:

[image]

in order to print the actual supports needed:

[image]

In this case, I'm printing PETG with extruder 1 and PVA for support on extruder 2. PVA can be started directly atop PETG (if the configuration is right) and does not need the portion of the support touching the build plate (which is a considerable amount). In other cases, e.g. tree support or extremely low density support, the same material can even be used for the primary and support extrusion, in which case it's obviously not required to touch the build plate.

fieldOfView, 2018-08-19:
You seem to (understandbly) misunderstand the support placement setting. It is about overhangs that are either directly over the buildplate, or over an already printed part. With "touching buildplate only", only overhangs that are directly over the buildplate are supported, so as to not create support material on top of an already printed surface (and possibly scarring that). This setting has nothing to do with connecting supports to the buildplate.

What you are seeing is caused by the setting "Support Horizontal Expansion". In the standard PVA profile, this is set to a non-0 value. If you set it back to 0, support will only go straight down. Use the search on top of the custom settings to view the setting, and try different values (0, a small value and a larger value) to see what the setting does.

mqudsi, 2018-08-19:
Thank you very much.

closed 2018-08-19 by mqudsi