Ultimaker/Cura #4258: section of part missing. changes with rotating part

supersoulman, 2018-08-19:
**Application Version**
3.4.1
**Platform**
Mac OS 10.13.6

**Printer**
PRUSA i3 MK3

**Steps to Reproduce**

ADD PART TO BUILD PLATE. SECTION OF PART IS MISSING. 

**Actual Results**
THE SECTION THAT IS MISSING CHANGES IF YOU ROTATE THE PART

**Expected results**
THE FULL PART SHOULD BE THERE

**Additional Information**
NONE
[image]
[image]

smartavionics, 2018-08-19:
Please provide the model file. Even better, please zip the cura project file and attach that to this issue. Thanks.

fieldOfView, 2018-08-19:
This is 90% certain caused by a "non manifold" model. Check the model in X-ray view, and you should see red parts. Best thing is to repair the model before printing. 

You may try if it slices better with the setting "Union Overlapping Volumes" turned off (use the search on top of the Custom settings).

BagelOrb, 2018-08-23:
I would wager it is 99% certain.

closed 2018-08-23 by BagelOrb