Ultimaker/Cura #4259: Request : Tool Change Enhancement

Weruminger, 2018-08-19:
**1. Fan control**
It would be helpfull to set "P" parameter for G-Code "M106" to the actual used Tool Index by CURA:
sample :
```
; some gcode ...
T0 
; some gcode ...
M106 P0 S127
; some gcode ...
M106 P0 S0 ; this can be set in Machine settings in extruder0 "End" code  
T1 
M106 P1 S127 ; current fan speed value can be set in Machine settings in extruder1 "Start" code
; some gcode ...
M106 P1 S255 
: 
; end of gcode ...
M106 P0 S0 ; this can be set in Machine settings in machine "End" code  
M106 P1 S0 ; this can be set in Machine settings in machine "End" code  
```
**2. Machine Retraction**
Using G10, G11 and G10 S1, G11 S1 for retraction and tool change retraction will produce a more universal g-code (e.g. running on equal machines with different tool change technique.
also the g-code becomes smaller and more readability or post processing possibility. 

**Application Version**
CURA 3.4.1 
request for future version

**Platform**
Windows 10 / X64
request is platform independend (g-code engine)

**Printer**
DIY I3 Mendel based on CTC I3 Clone 
upgraded with a pure magnetic tool chain machanism (w/o magnetic solenoid)
[Construction](https://github.com/Weruminger/Magnetic_Parking_Extruder.git)
request is usefull for any printer with a tool change upgrade
[MPE_TEST_Color2.zip](https://github.com/Ultimaker/Cura/files/2300264/MPE_TEST_Color2.zip)

**Steps to Reproduce**
NONE

**Actual Results**
NONE

**Expected results**
NONE

**Additional Information**

smartavionics, 2018-08-19:
This is related to something I did a little while ago which was to be able to specify the M106/M107 P value as a setting. It didn't get merged into cura because most printers are happy with the default value of 0. If I made that setting capable of being specified per-extruder, would that do what you want?

smartavionics, 2018-08-19:
Just to clarify - you don't want the tool index to be used for the M10[6/7] P value because machines that have multiple extruders but only one print cooling fan want to use the same P value for all tool indices. So you need to have the print cooling fan index as a setting that can be set per extruder and which has a default value of 0 (for compatibility with how it works now). Agreed?

Weruminger, 2018-08-20:
What i mean is something like a boolean switch in the Settings view, that enable "CooleIndexFollowsTool" and than set M10[6/7] P value to the extruder index and if not set leave it empty. 
In this case every combination:
- Separate Build Colling Fan for any Extruder 
- One for all 
is applicable. 
On the other hand it is easy to post process "M10[67] P\d" -> "M10[6/7] than reverse, because you get the String by layer in "Post Processing" Plugin and not by line what increase the script complexity a bit to find out and store the current tool index.

It is the same with retraction. G10 / G11 is easy to identify in "Post Processing" Plugin but hard to generate fail free from G-Code G1 E moves. 

so it is just a hint but may improve CURA output. I found out CURA as a very good tool and main "working horse" for slicing. ( i just do not find the menu point to manualy connect and disconnect to a connected printer ;-) )

And the Post Processing Plugin using Python Scrips is a great enhancement for easy G-Code Manipulation.

smartavionics, 2018-08-20:
> What i mean is something like a boolean switch in the Settings view, that enable "CooleIndexFollowsTool" and than set M10[6/7] P value to the extruder index and if not set leave it empty.

Most printers have fan 1 mapped to the hotend cooling fan so you would not want to use fan number 1 as a print cooling fan. I assume you have a different setup if you can use fan numbers 0 and 1 as print cooling fans.

I realise that having a simple checkbox to enable using the tool index as the P value to M10[67] is convenient but I am concerned that would make it too easy for someone to control the hotend fan when they thought they were controlling the print cooling fan and this could lead to damage to their hotend.

Therefore, I would rather have an explicit fan number specified per extruder which defaults to 0 for backward compatibility.

The G10/11 S1 parameter specifies a long retract? I am not familiar with it as I don't use firmware retraction. So there could be a checkbox along with the firmware retract setting that let's the user choose long/short retracts?

Weruminger, 2018-08-20:
No this is handeled by the FW and works fine. it is worse to send a wrong P value than none.

Weruminger, 2018-08-20:
sorry i mean Yes it is h......

smartavionics, 2018-08-20:
So, please tell me a little more about using the S param to G10/G11. When do you want the S param added and how is the value decided?

Weruminger, 2018-08-20:
to be precisely.
implicit Fan control 
is supported by Smoothie ware
is for now not supported by marlin BugFix 2.0.x
i did not check other firmware

Weruminger, 2018-08-20:
G10 / G11 with S1 (the is no other value supported) applies the behavior of Tool/Filament change retraction.
that means it does not only retract the specific among of filament it also recognize if the current selected tool/filament is already retracted and does prevent multiple G11 S1 retraction on the same tool/filament and on the other hand it just resume if the current Tool/filament is already retracted.

so the firmware manage the complete retraction process for multiple tools/filaments.

smartavionics, 2018-08-20:
>to be precisely.
implicit Fan control
is supported by Smoothie ware
is for now not supported by marlin BugFix 2.0.x
i did not check other firmware

http://marlinfw.org/docs/configuration/configuration.html#part-cooling-fan-multiplexer

Weruminger, 2018-08-21:
yes that i have Seen also before. This ist a nice solution. And you can integrate a specific electronic for it ... it is a multiplexer Signal to multiplex  one Pam Signal to multiple blower... not what I asked for. I did program the solution into my marlin fork on my self and this works fine as expected. 
Flag := false
Every thing works as before.
Flat := true
M10[67] Px  Sy. Set the blower for extruder x to value y
M10[67] Sy.        Set the blower for active extruder to value y

As described in GCode documentation an Backward compatibel

But it would be fine if curator will support the gcode generation as described in the gcode documentation

For using G10, G11, G10 S1, G11 S1 from slicer side there is no clear working solution G29 implementation is experimental and does not work. And to set Firmware retraction guessing by gcode is not a proper solution in my eyes.

smartavionics, 2018-08-21:
OK, well I have produced a Cura mod that allows you to specify the P value for M10[67] commands. I also provide a means to set it within the extruder settings like this:

[image]

I will submit PR's and the Cura devs can decide if they want to incorporate them or not.

Weruminger, 2018-08-21:
That‘s great. 
This looks exactly for what I expected. I am sure the g-code will cover the requirement.
Thanks a lot for your support and the time.

nallath, 2018-10-03:
But how can we confirm that suddenly adding the extra P won't mess up the code for certain machines? I'm a bit worried about the whole "Should work" statement earlier.

smartavionics, 2018-10-03:
The P only gets added when the fan number is non-zero.

Vandrasc, 2018-10-19:
This will be added in the next release, thank you!

closed 2018-10-19 by Vandrasc