Ultimaker/Cura #4260: Missing temperature settings

BillyYap, 2018-08-19:
**Application Version** 3.4.1

**Platform** Mac

**Printer** Anycubic Kossel Linear Plus

**Steps to Reproduce** Not sure, was working fine at first.

**Actual Results** Temperature settings from "Material" tabs are missing when i start the program. Tried changing to another material using the drop down doesn't help. Nor did de/select the temperature option in visibility selection helps. After i generate the Gcode, temperature setting were not in the Gcode as well. M190, M104 & M109 is not included in the Gcode 

**Expected results**

**Additional Information**

fieldOfView, 2018-08-19:
Please check that the gcode flavor in Machine Settings is not set to "Ultimaker 2"

BillyYap, 2018-08-19:
Ops. thanks for the reminder. my bad...

closed 2018-08-19 by BillyYap