Ultimaker/Cura #4262: Cura hangs when saving if multiple stl loaded

Psychokiller1888, 2018-08-20:
**Application Version**
3.4.1

**Platform**
Windows 10, x64

**Printer**
Vertex Delta custom added
I can't attach a file, it crashes when saving too

**Steps to Reproduce**
Load any stl
Load another stl
Try to save to file after slicing

**Actual Results**
Cura goes unresponsive, nothing happens, force closing as not responding is needed

**Expected results**
Well, actually save

**Additional Information**
I have try with and without auto slicing. I have removed statistic sharing as it was said on another bug report it could be it

[stderr.log](https://github.com/Ultimaker/Cura/files/2301599/stderr.log)
[cura.log](https://github.com/Ultimaker/Cura/files/2301600/cura.log)

alekseisasin, 2018-08-21:
@Psychokiller1888 , it is very strange that you have a certain issue.  I cannot reproduce this issue according to you "Steps to Reproduce"
Can you add any other printer, like Ultimaker 3, and reproduce these steps. And reproduce these steps without connection to the 3D printer. ( unplug the USB cable from your 3D printer to your PC)

If the issue occurs each time, you can try to remove Cura setting files from user folder and reinstall Cura. Before it, you can archive those files and upload them here. ( they do not contain any private information, only Cura settings). Then we can investigate further.

Psychokiller1888, 2018-08-21:
Hey, thanks for the reply! I am not connecting to a printer directly, I print on my own cloud through Octoprint. Will try that and share the settings!

Psychokiller1888, 2018-08-21:
[cura.zip](https://github.com/Ultimaker/Cura/files/2307150/cura.zip)

Ok, totally uninstalled Cura, manually removed everything related remaining like configurations in user data  and reinstalled... It's the same, exactly. Open one object, you can save the slice and/or the project. Open another object in the same, trying to save the slice or the project and cura crashes. Same if you load an object, slice it, save the slice, then delete it, and try to import another object, either it crashes directly when clicking the "open" left bar button (95% of the cases) or will crash when trying to save

Btw, the 3rd-party tag is not correct, this is raw Cura

alekseisasin, 2018-08-22:
The 3rd-party, it means that you are using not Ultimaker 3Dprinter.
Ok, what we can do else is to check "Event Viewer" because in Cura I do not see Error messages. The Event Viewer should provide additional information.

"_manually removed everything related remaining like configurations in user data and reinstalled... It's the same, exactly._"
     You tried without any connection to the printer. I want to be sure that connection to the 3Dprinter does not lead to this issue.

We do not have a debug argument, in case of the crash you should see the crash dialog. I assume it does not pop up. Am I right?

Psychokiller1888, 2018-08-22:
Nope, just freezes, goes unresponsive, and after clicking 10 times on the unresponsive top right close button, Windows quicks and and proposes to force close it

I do not have direct printer connection

Psychokiller1888, 2018-08-22:
This is opening after slicing bug: https://puu.sh/BinBw.mp4

This is double model import bug: https://puu.sh/BinDE.mp4

Psychokiller1888, 2018-08-22:
Event report:

> Détecteur d'erreurs 2214116884309123866, type 5
> Nom d’événement : AppHangB1
> Réponse : Non disponible
> ID de CAB : 0
> 
> Signature du problème : 
> P1 : Cura.exe
> P2 : 0.0.0.0
> P3 : 5b113154
> P4 : 6a87
> P5 : 134218252
> P6 : 
> P7 : 
> P8 : 
> P9 : 
> P10 : 
> 
> Fichiers joints :
> \\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WER94FA.tmp.WERInternalMetadata.xml
> \\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WER951B.tmp.xml
> WPR_initiated_DiagTrackMiniLogger_OneTrace User Logger 20180820 Event Collector_0_inject.etl
> \\?\C:\Users\Psychokiller1888\AppData\Local\Temp\WER956A.tmp.etl
> WPR_initiated_DiagTrackMiniLogger_WPR System Collector_inject.etl
> \\?\C:\Users\Psychokiller1888\AppData\Local\Temp\WER956B.tmp.etl
> \\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WER9568.tmp.csv
> \\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WER9588.tmp.txt
> \\?\C:\Users\Psychokiller1888\AppData\Local\Temp\WER9BD4.tmp.appcompat.txt
> 
> Ces fichiers sont peut-être disponibles ici :
> \\?\C:\ProgramData\Microsoft\Windows\WER\ReportArchive\AppHang_Cura.exe_d84b1cecfddff5ee1693b24fbf5d05e629d433f_27e06e5a_1fa59f7a
> 
> Symbole d’analyse : 
> Nouvelle recherche de la solution : 0
> ID de rapport : e7e1767d-8703-4659-9603-e8f862f7d822
> Statut du rapport : 268435456
> Récipient avec hachage : 2460d8de02bf939a7eba1f934c70a31a
> GUID CAB :0

> Le programme Cura.exe version 0.0.0.0 a cessé d'interagir avec Windows et a été fermé. Pour voir si plus d'informations sur le problème sont disponibles, vérifiez l'historique des problèmes dans le Panneau de configuration Sécurité et maintenance.
>  ID de processus : 2c74
>  Heure de début : 01d439ed7d04ecd4
>  Heure d'arrêt : 9
>  Chemin d'accès à l'application : C:\Program Files\Ultimaker Cura 3.4\Cura.exe
>  ID de rapport : e7e1767d-8703-4659-9603-e8f862f7d822
>  Nom complet du package défectueux : 
>  ID de l'application relative à un package défectueux : 
>  Type de blocage : Cross-thread

Looks like a multi thread problem

alekseisasin, 2018-08-22:
I also assume it is multithreading problem, maybe deadlock, but why you have it and I do not.
We can try the next scenarios:
1 - Turn off internet connection and repeat "Steps to Reproduce".
2 - Run as Administrator and repeat "Steps to Reproduce".
3 - If you have antivirus turn off it and repeat "Steps to Reproduce".

alekseisasin, 2018-08-22:
I see that you are loading models from "windows Onedrive"
Can you load models from Local Drive C:\?

Psychokiller1888, 2018-08-22:
Good catch! Will try this! And also the three other steps. It's on onedrive because working from 3 different computers...

Psychokiller1888, 2018-08-22:
Files directly on C: = same
Internet off: same
Kaspersky off: same
Run as admin: same

All the above points were cumulated in the order I wrote them

Stdout

alekseisasin, 2018-08-23:
There are maybe some programs which are build using QT framework. I think you also installed some graphic design programs, they might conflict. 
I was thinking do you have enough free space on your local drive? :)

Psychokiller1888, 2018-08-23:
Largely :)

[image]

Graphic programs? I use Blender but blender open or closed, the same happens to Cura

We need a stacktrace if we want to hunt down that bug

alekseisasin, 2018-08-23:
I see you have Photoshop, I do not think it conflicts with it.
I want to share NOT_STABLE Cura version, it has a new QT framework version and maybe the problem will be gone for you.

https://wetransfer.com/downloads/3b7a298ae6ee53070ffe7dc01a9bb95e20180823105012/4b72789ab102ddda35e4d6535fa9a1bd20180823105012/1a0b1b

It is not stable, but you can try it. It is our master build(latest),

Psychokiller1888, 2018-08-23:
Ok, downloading, thank you. Will be back with news soon :)

fieldOfView, 2018-08-23:
Best make a backup of your configuration folder before you run that build. There may be issues going back to the stable build (or foreward to the next official release)

Psychokiller1888, 2018-08-23:
yep, did, installing now

Psychokiller1888, 2018-08-23:
Aaaaaand.... Same happens....

alekseisasin, 2018-08-23:
I curious what computer do you have. What computer(configurations) do you have?

Psychokiller1888, 2018-08-23:
Windows 10 1803/17738.1000

Asus Maximus Hero VIII, I7 Skylake 6700K, 32 Go DDR4 @3200mhz Corsair Ram, Asus Geforce 1080TI OC edition 11go ram, samsung evo 840 and 850 for ssd and a caviar black WD harddrive, dual screen, htc vive plugged in, wired network, no overclocking

Psychokiller1888, 2018-08-23:
This is really a pain...... Can't print more than one piece. I have to use repetier Vertex Delta custom slicer that is really bad compared to your awesome Cura. I'll have a look at the code and add some flags when I can

Psychokiller1888, 2018-08-23:
I just tried to close EVERYTHING I have running (define fancy :) ) (google chrome, jetbrain pycharm, steam, steam background etc etc etc) And it still does the same

alekseisasin, 2018-08-23:
This is very strange. Before we had problems with the UI, some latest video cards did not work correctly with Cura, for this reason, we asked to update the video card drivers. I am not sure, can you check the drivers(video card, motherboard, etc..)? So far I cannot realize what leads to this issue.

fieldOfView, 2018-08-23:
What GPU does Cura use on your system? The 1080TI or the integrated Intel® HD Graphics 530? Can you force it to use the other?

Psychokiller1888, 2018-08-24:
I did not check that tbh, so it uses the default one, I guess Cura defaults to integrated? I will check that asap, at work atm...

My drivers are up to date, Geforce Experience even updated the drivers 2 days ago. As for the rest they also should be. Bios is from about a month or 2 ago

Psychokiller1888, 2018-08-24:
On the 1080TI:

[image]

Psychokiller1888, 2018-09-01:
So, no ideas how to go forward? It's really a pain... I have to group my models in blender, export them and then load the group in Cura...

Since the bug has been declared my windows has updated and my graphic card as well

tachikoma1373, 2018-09-09:
I have a very similar problems

I don't use SD card, I use octoprint by saving gcode and uploading. system is  i5, 8GB RAM, Asus Mobo and RX580, 250GB SSD + 1TB storage all SATA.  All drivers are WHQL certified latest versions. Here's an extract from the cura log - loaded up 3 models, sliced, save gcode and BANG.

2018-09-09 12:18:20,516 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onSlicingFinishedMessage [611]: Slicing took 2.4562132358551025 seconds
2018-09-09 12:18:20,517 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onSlicingFinishedMessage [612]: Number of models per buildplate: {0: 3}
2018-09-09 12:18:20,517 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onSlicingFinishedMessage [626]: See if there is more to slice...
2018-09-09 12:18:20,519 - DEBUG - [MainThread] UM.Backend.Backend._logSocketState [185]: Socket state changed to Closing
2018-09-09 12:18:20,520 - DEBUG - [MainThread] UM.Backend.Backend._onSocketError [207]: Socket debug: Arcus Error (13): Closing socket because other side requested close.
2018-09-09 12:18:20,521 - DEBUG - [MainThread] UM.Backend.Backend._logSocketState [187]: Socket state changed to Closed
2018-09-09 12:18:20,533 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onBackendQuit [765]: Backend quit with return code 0. Resetting process and socket.

as you can see its the cura backend that fails. I cant see how its to do wit gcode rendering when the slice clearly completes.  surely all its doing having done this is saving a file(?)  and thats where it dies.

Here' the windows EVT log error:

The program Cura.exe version 0.0.0.0 stopped interacting with Windows and was closed. To see if more information about the problem is available, check the problem history in the Security and Maintenance control panel.
 Process ID: 3174
 Start Time: 01d44827fe3e7954
 Termination Time: 5
 Application Path: D:\Program Files\Ultimaker Cura 3.4\Cura.exe
 Report Id: 6385479b-3d21-42ff-9a76-ef89da3be70f
 Faulting package full name: 
 Faulting package-relative application ID: 
 Hang type: Cross-thread

This happens 9/10 times with multiple models and 5/10 times with single models.

Psychokiller1888, 2018-09-14:
At least you have a log!

nallath, 2018-09-14:
Nothing in the log indicates that it's the engine that fails. Every time you slice, the engine is re-started. What you are seeing is exactly that; The engine is done slicing, it sent the data back and it shuts itself down.

Rouzic6666, 2018-09-22:
Hello,
I notice the same behavior in cura 3.4.1. I have the following configuration:
 <UM2>--<USB>--<Repetier-Server-Free0.90.1>--<EthernetLAN>--<LinuxMint(cura 3.4.1)>
When I press connect in cura it crashes

Psychokiller1888, 2018-09-23:
Cura is now unusable for me, stops working whenever I hit "Save". Sad, it's a good soft, but there's no try at fixing this, I'll check other slicers

Psychokiller1888, 2018-10-04:
Latest 3.5 update seems to have fixed the issue for me

Ghostkeeper, 2018-11-29:
We still don't know exactly what happened here, but I suspect it has something to do with multi-build-plate slicing. I'm glad that the problem disappeared...

closed 2018-11-29 by Ghostkeeper

Rouzic6666, 2018-11-29:
I gave up Repetier-Serve and switch to Octoprint. Cura 3.5.1, and now 3.6, work perfectly!