Ultimaker/Cura #4264: Dual Extrusion slicing problem

martinnmayer, 2018-08-20:
**Application Version**
Cura v.3.4.1
**Platform**
Windows 10
**Printer**
Createbot MID

**Steps to Reproduce**
Before stating the steps, I want ti describe what I'm trying to achieve. I started using PVA in the second extruder for model supports, but because of cost issues, I'm trying to make all supports in PLA and switch to PVA just before it touches the actual model. In order to do so I did the following (starting from the default "Fine" settings.

1. Enable generate support
2. Set support extruder to 2 (PVA)
   2.1 Set support infill extruder to 1 (PLA)
   2.2 Set support roof extruder to 2 (PVA)
   2.3 Enable support roof (this seems to be necesary for the previous settings to apply).

**Actual Results**
After slicing everything seems to be ok except for the machine travels, which are a little erratic and suboptimal. Whats most important is that it makes more than one travel to [0;0;Z] and hits the corner. After moving back to the print, the hit caused and offset which affects the following layers.
This reminds me of other machines that need to do a tool change.

**Expected results**
Generate supports with PLA infill and PVA contact layers with model.

**Additional Information**
In the following screenshot, you can see what I'm trying to achieve:
[image]

In this screenshot, you can see the travels:
[image]

I'm really sorry if this report does not follow the guidelines, I followed along the template paying special atention to doing so.

smartavionics, 2018-08-20:
This may be fixed for a future release by https://github.com/Ultimaker/CuraEngine/pull/776.

BagelOrb, 2018-08-23:
In order to assist you:

Please attach your .curaproject.3mf file.

Also please give an exact example of what is wrong. Take a screenshot of one layer and draw an arrow to where the problem is. Or copy part of the gcode commands which are the problem.

Ghostkeeper, 2018-08-31:
Good bug report. It's indeed probably fixed by Ultimaker/CuraEngine#776 for our next release. To confirm that we'll need to have the model or preferably the project file that caused this problem. Could you go to File -> Save Project... and attach to a comment here the file that you get from there (change the extension to .zip so that Github accepts it).

Ghostkeeper, 2018-10-25:
Let's assume it's fixed by CuraEngine#776 then, for lack of better information.

closed 2018-10-25 by Ghostkeeper