Ultimaker/Cura #4269: Entire application is useless garbage

zinenu, 2018-08-22:
**Application Version**
ALL

**Platform**
ALL

**Printer**
ALL

**Steps to Reproduce**
USE THE SOFTWARE TO DO LITERALLY ANYTHING

**Actual Results**
CRASHES, BUGS

**Expected results**
ABILITY TO HANDLE A THERMAL RUNAWAY OR ABORTED PRINT

**Additional Information**
CURA IS FUCKING TRASH AND YOU SHOULD FEEL BAD FOR MAKING IT, JUST QUIT. LITERALLY ANY OTHER SLICER IS BETTER.

nallath, 2018-08-22:
[image]
Just because it's cruise control, doesn't mean you don't have to steer. 

That being said; Cura can handle aborts and thermal runaway is the firmware of the printer that is messing up.

closed 2018-08-22 by nallath

ChrisTerBeke, 2018-08-22:
Thanks for this positive feedback. We will most certainly look into it.

ChrisTerBeke, 2018-08-24:
Don't wanna make this bigger than it is, but (stock) Marlin has plenty of issues with thermal runaway, just Google for it.