Ultimaker/Cura #4271: v3.4.1 picking up old deleted printer details from somewhere

robinbon, 2018-08-22:
Having updated to 3.4.1

Now each time I load app it says configuration error on a printer I have not had for a year.

This printer was deleted several updates ago last year.

[image]

fieldOfView, 2018-08-22:
An option to remove only the offending files instead of nuking the whole configuration would be nice.

BagelOrb, 2018-08-23:
I think it's better to just suggest this user to delete some of his files. 

Maybe the Cura team should prevent such failures in the first place, but managing the defects in a more user friendly way is not really at the top of the priority list if you ask me.

BagelOrb, 2018-08-27:
What operating system do you have?

robinbon, 2018-08-27:
mac OS, latest

BagelOrb, 2018-08-28:
This should help you further: https://github.com/Ultimaker/Cura/wiki/Cura-Preferences-and-Settings-Locations

diegopradogesto, 2018-10-02:
Originally this Configuration Errors were created to detect specific cases like this and automatically delete the files. However this may cause undesired deletion of some important files and then make the application totally unusable. We then decided just to warn the user that some files may be corrupted.

In your case, it was probably caused by some but in the past versions and those files were orphans of that printer you deleted. Now you see these errors. The best option is to either start from a clean install and start configuring again your environment or go to the config folder and manually check if those files are not linked to any other and safely delete them.

This issue was created some time ago, so maybe you already manage to clean up your config folder. Is it something we can help you with? Let us know.

Ghostkeeper, 2018-11-26:
We considered the option to delete only the broken files, but the problem is that this tends to cascade into other files that depend on the broken files, so those will then give a problem next time.

My preferred solution (after preventing this bug in the first place) would be to show this error only once per file.

nallath, 2019-01-07:
Closing due to inactivity

closed 2019-01-07 by nallath