Ultimaker/Cura #4272: Start Gcode - T{initial_extruder_nr}

amagr0, 2018-08-22:
**Application Version** - CURA 3.4.1
**Platform** - Win 10 Pro, 64 bit
**Display Driver** - NVIDIA GeForce 840M

**Steps to Reproduce** - Change Start Gcode to:

G28
T{initial_extruder_nr}
M104 S{material_print_temperature}
M109 S{material_print_temperature} 

**Actual Results** 

;Generated with Cura_SteamEngine 3.4.1
T0
G28
T1
M104 S200
M109 S200

**Expected results**

;Generated with Cura_SteamEngine 3.4.1
G28
T1
M104 S200
M109 S200

Is there any way do avoid the T{initial_extruder_nr} placed by Cura at the beguining of the start gcode.
I need to do a G28 before the T{initial_extruder_nr} to avoid a little shock with the extruder on the frame of he printer.

Thanks

BagelOrb, 2018-08-23:
There is currently no setting to adjust that behaviour.

Ultimaker doesn't have time to implement this for you, but you can give it a go yourself.

fieldOfView, 2018-08-23:
The easiest solution would be to use a postprocessing script.

I have not tested this, but I think you could use the Search and Replace one to search for ";Generated with Cura_SteamEngine 3.4.1\nT1" and replace it with ";Generated with Cura_SteamEngine 3.4.1".

amagr0, 2018-08-23:
Hello @fieldOfView, thanks for your suggestion, unfortunately it does not work.

@BagelOrb, this implementation is not only for me, is for all with non Ultimaker multi extruders printers, but I understand the position of Ultimaker.

BagelOrb, 2018-08-23:
There are quite some issues with the order of the initial couple of commands. Pull Requests are welcome, but we wouldn't want 5 settings just to decide on the order of the first couple of commands. We need a principled way of dealing with this. Somebody needs to think this through before he presents a solution for Cura.

fieldOfView, 2018-08-23:
I think you have to tick the "Use regular expressions" box for this to work (because of the \n)

fieldOfView, 2018-08-23:
Better search pattern: `;Generated with Cura_SteamEngine.*?\nT1` with the "Use regular expressions" box ticked.

amagr0, 2018-08-23:
Hello @fieldOfView thanks for your suggestion, but does not work.

@BagelOrb this is implementation is not for me, is for all the community that have (non Ultimaker) multi extruders printers, but I understand the position of Ultimaker.

amagr0, 2018-08-23:
Thanks @fieldOfView it worked!

EmJay276, 2023-07-24:
Here is a even better regex search pattern
Search: `;Generated with Cura_SteamEngine (.*?)(\nT[0-9])`
Replace: `;Generated with Cura_SteamEngine \1`

This keeps the Cura_SteamEngine version

GregValiant, 2026-09-06:
I'm cleaning up some of these old reports.  I'll close this.  Currently 5.13.0 is the latest Cura version.
BTW, The "Initial Extruder Number" has been fixed and broken a few times.  It is currently (5.13.0) working correctly, but there were side effects.  It will need more tweaking.

closed 2026-09-06 by GregValiant