Ultimaker/Cura #4274: Support placement setting is blank and "settings updated" messege appears. Cura 3.4.1

printingotb, 2018-08-22:
**Application Version**
3.4.1

**Platform**
Win7

**Printer**
At least on custom FDM printer

**Steps to Reproduce**
I am not sure when this happens, last time was when I discarded current changes from my profile. The changes had nothing to do with support settings.
It happens on different profiles and machines.

**Actual Results**
I get the message "Settings Updated" settings have been changed to match the current availability of extruders: [Support type]
And the support placement setting has become blank.
The star that indicates that there are changes to the profiles does not disappear when discarding changes and the support settings is still blank.

**Expected results**
My profile should remember the support placement setting and not suddenly loose the value.

**Additional Information**
[image]
Project file:
[Support placement blank.zip](https://github.com/Ultimaker/Cura/files/2310377
/Support.placement.blank.zip)

Ghostkeeper, 2018-10-24:
I'm unable to reproduce this so far. Can you try with Cura 3.5.1? Maybe it's been fixed since you reported this.

printingotb, 2018-10-24:
I have been running 3.5.1 for a while. I am actually not sure if it is still an issue. I had it a few days ago but I could have been when I ran 3.4.1.
I'll keep my eyes open for a while and if it doesn't re-appear, we can close this.

printingotb, 2018-10-24:
Interestingly it says "buildplate" when viewed from the manage profiles menu.
[image]

Ghostkeeper, 2019-01-02:
> If you open this project, do you get the issue as well?
Untitled.zip

This is a completely empty file. It was saved as normal 3MF instead of as Cura project. Or at least, it appears to be. The `cura` folder is missing from the 3MF's zip archive.

> Interestingly it says "buildplate" when viewed from the manage profiles menu.

This is expected behaviour. If it said `0` then that might have been the bug.

printingotb, 2019-01-02:
I must have made a mistake when saving that. The thing is that it shows up as blank in the settings in the main window but if entering the profiles menu it says buildplate.

I have had the issue a few more times but it seems really rare and not a big deal, I just need to choose a support setting and update the settings profile again. 
Once 4.0 is out of beta I will make a clean install and merge all my old profiles from different Cura versions (I had to go back to 3.4.1 for Ironing and stupidly worked on some profiles in 3.5 and some in 3.4.1).

JalteJKR, 2019-05-23:
I have the same problem, please watch my video https://youtu.be/vp-tibeVCjM

printingotb, 2019-05-23:
I'm not alone :D
Nice video. I hadn't realized that the issue appeared every time I changed a support setting and then choose discard settings! 
The issue goes away if I first update the profile with support placement to any of the options and then turn it off and update the profile again.

JalteJKR, 2019-05-24:
Hmm, doesn't work for me :( I'm probably just going to copy the settings from my current profile to the default one that works :/

Ghostkeeper, 2019-05-28:
Thanks for that video. It helped to deduce where this problem came from. I made a fix here: https://github.com/Ultimaker/Cura/pull/5795

I don't dare to make this change to master without letting it be reviewed by my colleagues so I'll discuss it tomorrow with them.

PR #5795 (2019-05-31): Fix auto-resetting settings when discarding user changes
This fixes a bug where settings are being reset to "0" (or whatever the default extruder number is) when discarding current changes to settings. This happens to all settings that are:
* In the quality-changes profile.
* Not currently visible.

Why would it need to reset all invisible settings? It shouldn't matter whether a setting is visible or not.
This is causing bugs because the settings that are being reset are not necessarily extruder-type settings (as that is checked below). They are being reset to the value self._default_extruder, [... cut]
commit 63cd3a52 (2019-05-31): Don't reset all invisible settings

Why would you need to reset all invisible settings? It shouldn't matter whether a setting is visible or not.
This is causing bugs because the settings that are being reset are not necessarily extruder-type settings (as that is checked below). They are being reset to the value self._default_extruder, which is not always a valid setting value.

Fixes #4274.
closed 2019-05-31 by LipuFei