Ultimaker/Cura #4276: File name not saved

Lizartin, 2018-08-23:
**Application Version**

**Platform**

**Printer**

**Steps to Reproduce**

**Actual Results**

**Expected results**

**Additional Information**

ChrisTerBeke, 2018-08-24:
Please fill out the issue template properly, this is not enough information.

CCS86, 2018-09-19:
Same issue here.

3.4.1 / Windows 10 x64 / multiple printers

Click to edit file name. Change name. Click save-> original name is saved. Trigger a settings change that causes reslicing->displayed filename reverts to original, automatic file name.

fieldOfView, 2018-09-19:
Could you try the 3.5 beta? I remember fixing this.

CCS86, 2018-09-19:
I just did a little test. This issue was occuring for me only during 'save to removable drive'. I switched to 'save to file' and in the windows save dialog, the edited file name was correctly displayed. I cancelled, then switched back to removable drive, and it saved with the correct edited file name.

Lizartin, 2018-09-19:
Hi,
I almost always just save to "removable file".  This just started happening with one of the recent updates.
Liz

CCS86, 2018-09-19:
Another issue I see is that after the little trick above to allow saving directly to a removable drive, the file name gets truncated if it contains a period. For example:

"ConcreteMold0.8_2" gets saved as "ConcreteMold0.gcode"

fg89o, 2018-09-22:
> Could you try the 3.5 beta? I remember fixing this.

In 3.5 beta is working fine. Thanks.

closed 2018-09-27 by nallath

fieldOfView, 2018-09-27:
@Ccs86, when you find another bug, please open another issue or it will get lost.

CCS86, 2018-10-04:
I just installed 3.5 final and the bug is still present. 

Actually, even the 'save as file' workaround I was using won't work now:

[image]