Ultimaker/Cura #4277: Feature request: single body print with multiple extruders

mqudsi, 2018-08-23:
Now that the Ultimaker 3 finally has access to print cores ranging from 0.25 to 0.8, it would be awesome if it were possible to print a single part with both extruders, one for finer details and the other for the bulk of the part with the same material loaded into both cores.

For the sake of simplicity, if we define "finer details" as components of the model present in the original STL (to avoid having support components or infill being dual extruded), that would mean that portions (or layers?) of the design that cannot be represented with the larger print core are extruded with the smaller diameter print core, automatically.

diegopradogesto, 2018-10-02:
I would say that it's difficult to automatically determinate which parts should be printed with "fine detail" in the same layer. However Cura allows the user to use multiple options to tweak every print. I would suggest:

- Start using _Adaptive Layer Height_ for fine layers.
- Use per-extruder-settings (such as printing the infill with 0.8 nozzle and walls with 0.25)
- Use per-object-settings (you can overlap a single cube in a part that needs more accuracy and define this part to be printed with the 0.25 nozzle).

Ghostkeeper, 2018-11-26:
As a partial solution, you can set the inner wall, skin and infill extruder to be printed with the extruder with a 0.8mm nozzle, and the outer wall to be printed with an extruder with a 0.4mm nozzle (or 0.25mm but then you'll have no profile). This achieves more or less what you need.