Ultimaker/Cura #4285: Can i Restart de job From Layer Z (24)

SharkSteps, 2018-08-24:
Cura 3.4.1

Windows 10 Pro
CPU 2.8 Ghz Dual-Core, 8GB Memory, Geforce-105M
Printer - ANET-A8

Sometimes Cura stop responding in my computer i don't know if it was my computer or cura or even the printer.
Yesterday i solve a similar trouble, when i export from fusion the .STL file in High Refinement, and open with cura everything is OK, start printing few layers later the print stops in some position if i click on cura software i have a message cura stops responding, few seconds later he work normally but the printer doesn't continue, all menus working i can also send the printer to home.
If i Export the same .STL job From Fusion 360 In Medium Refinement everything works fine and flow.
But today i have the same problem again, without exporting in high, today the same append with the medium refinement, i have the 10h job in my printer plate, i know also the layer position she stop (24),
so my question is:
There anyway to restart the job from that layer.

My best regards guys, and you keep doing the good job

fieldOfView, 2018-08-24:
No, there is no easy way of doing that from Cura. The only way is to edit the gcode file by hand.

SharkSteps, 2018-08-24:
Thanks But how do i send the file after edit?, my cura doesn't open g-Code files?

SharkSteps, 2018-08-24:
Give me the 2 messages, 
"Invalid File 
 failed to load c:/users/....", i even try the original file i save from cura, 

The other message is :
G-Code Details
Parsing G-code and have a gray bar under

SharkSteps, 2018-08-24:
Here is the G-Code file
[PI3_Second_Box_Med.zip](https://github.com/Ultimaker/Cura/files/2319086/PI3_Second_Box_Med.zip)

fieldOfView, 2018-08-24:
I would need the Cura log to know why reading the gcode file fails. But I think it is a lot quicker and more reliable to just print those 24 layers again.

fieldOfView, 2018-08-24:
Ah, I see. In your start gcode, you use a deprecated {replacement_pattern}. Use {speed_travel} instead of {travel_speed}.

fieldOfView, 2018-08-24:
To get Cura to load the gcode file, replace {speed_travel} by the travel speed you set in Cura.

SharkSteps, 2018-08-24:
Yeah but if the problem come over again this 24 layers was 3h of printing, and where i can find the cura log?, the French guy has the same problem with G-code he can´t open. in last Post i just read in 5 languages but only to let you know what he say.

SharkSteps, 2018-08-24:
I now i solve the g-Code problem i have change the {speed_travel} in the printer profile, saved the g-code again, and finally i can open the g-code, so thanks for everything, now i will edit the g-code and try continue the job.

SharkSteps, 2018-08-24:
YESSSSSSS..... it WORK;S no problems Thanks again for everything.....

SharkSteps, 2018-08-24:
Working well i did start printing in layer 24 just edit the G-code file removed all 23 layers, change the layer 0 to 24 and the printing is running well.

closed 2018-08-24 by SharkSteps