Ultimaker/Cura #4286: probleme enregistrement cura

VolRed53, 2018-08-24:
bonjour,a chaque fois que je veut enregistrer un fichier gcode via cura , cura se coupe sans aucun message d'erreur et quand je veux l'ouvrir en cliquant sur l'icone du bureau il ne veut pas s'ouvrir.merci d'avance pour votre aide

fieldOfView, 2018-08-24:
Please fill out the issue template when creating issues here, and please use english.

SharkSteps, 2018-08-24:
VolRed53 Ils veulent que vous écriviez les messages en anglais et que vous vous conformez à ce que le modèle de la page vous indique, à savoir le logiciel CURA, la version Windows, le type de machine que vous utilisez et l’imprimante que vous utilisez. Si votre question est déjà posée en faisant une recherche sur le sujet.

VolRed53, 2018-08-24:
I use the version of cura 3.4.1, under Windows 10 professional with laptop for a printer Monoprice select mini V2

fieldOfView, 2018-08-24:
The Ultimaker forums have a section where the language is french, perhaps you can find help there:
https://community.ultimaker.com/forum/126-fran%C3%A7ais/

Here, we cannot help you if you cannot fill in the issue template in english, with information describing how to reproduce your problem.