Ultimaker/Cura #4287: Cura makes too many movements

dmitrygorbenko, 2018-08-24:
Hi,

Cura makes to many movements to print support structures.
I haven't checked, but maybe the same algorithm applies to a model itself.
To understand what I mean please take a loot at the image below:

[image]

Image number 1 is some initial state. On image 2 Cura moves head around model to print some support. It's OK.
But after Cura goes backward to the same place it has been a moment ago to print another support (see image 3).
This is what I'd like to change/improve.
After Cura prints next support - it's OK (image 4).
And on image 5 Cura makes another long movement where it was some time ago.

I think you've got the idea what I mean by "too many movements".

Can such movements be improved ?

Cura 3.4.1, Linux Ubuntu

[Titan Aero Carriage.curaproject.3mf.gz](https://github.com/Ultimaker/Cura/files/2319244/Titan.Aero.Carriage.curaproject.3mf.gz)

Thanks,
Dima

alekseisasin, 2018-08-28:
Hi @dmitrygorbenko, It is not a bug, it is intended behavior. This happens because of the setting _Combing Mode_,  please find _Combing Mode_ setting and set an option to "Off", then the extruder will go directly to the next printing area.
Also, in the next release, the unnecessary support will disappear(also related to your model) and we added a new option to the combing mode.

dmitrygorbenko, 2018-08-30:
I see, thank you for helping me !

closed 2018-08-30 by dmitrygorbenko

Ghostkeeper, 2018-10-26:
The reason it's making this movement is two-fold:

1. When checking which object to print next, we don't take collision avoidance into account. So it determines that the support bit on the other side is the closest one to print next, but in reality due to collision avoidance the path towards it is longer than some other parts.
2. When avoiding collisions, we don't check which parts of them have been printed yet and which are not. This is mostly for architectural reasons and performance reasons. It would be hard to implement efficiently. But also because even if you don't hit the current layer, it'll still hit the walls in the previous layer that you were trying to avoid.

dmitrygorbenko, 2018-11-07:
>> This is mostly for architectural reasons and performance reasons
I understand what it can be too hard to implement algo which takes into account collisions, but... it's kind of awful to see how your printer spends print time to move around the model.

I hope the management of Ultimaker/Cura product can find this issue important to fix.