Ultimaker/Cura #4289: [3.5 ppa] crash when connect to octoprint

redPanther, 2018-08-25:
**Application Version**
latest 3.5 master ppa

**Platform**
ubuntu 18.04

**Printer**
renkforce rf100

**Steps to Reproduce**
- have octoprint running
- open cura
- manage printers
- select printer
- press "connect to octoprint" and fill out the form
- enter the valid api key

**Actual Results**
application crashed with following console messages:
https://pastebin.com/FqG3sy2q

**Expected results**
no crash and enabled connection to octoprint

**Additional Information**
I encountered this behaviour in 3.4x too. in 3.3x it worked as expected

fieldOfView, 2018-08-26:
This was fixed in the master branch of OctoPrintPlugin 10 days ago.

ChrisTerBeke, 2018-08-26:
There is a 3.5 PPA? 3.5 is not even out yet....

closed 2018-08-26 by ChrisTerBeke

owallen, 2018-10-28:
I still get a crash when connecting to Octoprint. Same result in 3.5 and 3.5.1
Works fine on 3.4.1 on the same machine.
[cura.log](https://github.com/Ultimaker/Cura/files/2522964/cura.log)

fieldOfView, 2018-10-28:
@owallen The OctoPrintPlugin needs an update. You seem to be using OctoPrintPlugin 3.5.0. Your crash was fixed in OctoPrintPlugin 3.5.1.

owallen, 2018-10-29:
Thanks for your reply! I've installed the latest vesrion in Cura 3.5.1 Toolbox and the screen says 3.5.1 (see attached picture).
[image]

fieldOfView, 2018-10-29:
I am not very familiar with the .ppa, but could it be that you now have two versions of the plugin installed? I believe that there is a copy of OctoPrintPlugin in  cura-extra-plugins, which installs to a different location than when you use the Toolbox.

fieldOfView, 2018-10-29:
Sorry, OP was using the ppa, so I assumed you were too. My guess was correct though ;-)