Ultimaker/Cura #4290: Mac 3D Navigation

jflund, 2018-08-26:
**Application Version**
3.4.1

**Platform**
Mac  (High Sierra)

**Printer**
Ultimaker 3

**Steps to Reproduce**
Open any model and try to navigate around the model (zoom, rotate, etc.)

**Actual Results**
1. Inexplicably zoom is limited to a long-distance view unless a Camera Position is chosen first.
2. Rotating the build plate (using a two-finger click and move on the trackpad) rotates about a point that is equal to the distance between the camera and the center of the build plate that set when the Camera Position was selected. So if the camera has been moved toward the model, rotation occurs about a point behind the model (and behind the build plate, and behind representation of the printer). The result is that attempts to rotate cause the entire model to be rotated out of view. 

**Expected results**
1. The zoom should be allowed to move much closer to the model even if a Camera Position has not been chosen first.
2. Rotation should occur the point where the center of the viewport intersects the closest coordinate plane in a coordinate system with the origin at the center of the bounds of the the closest model to the camera. If the center of the viewport does not intersect with the model bounds, then a coordinate system where the origin is at the center of the build plate should be used. Find the intersection closest the camera between the coordinate planes centered inside of each model, or the build plate if no model is centered. Rotate about that point. In this scenario, I am calling the three planes defied as  x=0, y=0, and z=0 the "coordinate planes". I assume the coordinate plane remains oriented with x, y, and z axes parallel to the world x, y, and z coordinates.

**Additional Information**
Just make the camera movement natural. As of this latest release (3.4) it is highly difficult to use.

Ghostkeeper, 2018-10-25:
I'm unable to reproduce the first bug. Zooming works fine for me.

The second result requires some debate. I think the system you're proposing is more magical and less logical than our current solution. I think we have to keep it as simple as possible. If the point of rotation depends on the position of the models on the scene and their relation to the camera position and such, I think it will become very difficult to use. Especially if you have multi-extrusion prints where the meshes are wound around each other, or when multiple models intersect or have a surface Z-fighting in the same plane.

jflund, 2018-11-15:
Thank you for reading my comment. Maybe it was too complex, but I am honestly finding this unusable. So will you try to do something for me? I am not sure why others aren't having the same problem. Maybe I am not understanding, but I have used a lot of 3D apps, and this is the only one I find unusable. 

1) Will you open a file and click on the Top-Down view button to orient above the model. 
2) Then zoom in on an area of the model, any area, by sliding 2 fingers up on the trackpad with the cursor over that area of the model.
3) Now, pretend you want to inspect the front edge of the model while zoomed in. To do so you click and hold and slide two fingers up. Instead of rotating about the point on the build plate that the cursor is pointing to (I would argue the expected result), the view rotates about a point behind the build plate (I have figured out it is rotating about the spot that the build plate was immediately after clicking the top-down button.

The result is that instead of orbiting about your area of interest, the entire build plate flies off the top of the screen and becomes lost. It makes it impossible to zoom in on areas of the model that aren't printing properly and the looking around to see how it is supported, etc. I cannot believe that orbiting about a point way, way behind the build plate is the intended behavior. I can upload a video if you  can't reproduce it. 

I agree that it seems at first glance that orbiting differently depending on where a model is seems complex at first, but the result is that you actually orbit about the spot the mouse cursor is touching. This is extremely intuitive and is exactly how SketchUp works. 

Simplified recommendation:
However, SketchUp doesn't have a "build plate", so to keep your code simple, at a minimum I believe you should orbit about the plane the build plate is currently in, not about the plane it was in before zooming in on the model. To implement this you would only need to make a simple change in the order in which you multiply your view matrices from the way it is currently being done.

GregValiant, 2026-09-10:
I'll close this.  If the problem remains is 5.13.0 please fill out a new report.

closed 2026-09-10 by GregValiant