Ultimaker/Cura #4291: [3.4.1] Combining:all causes problems on round hole walls

jvdbrg, 2018-08-26:
**Application Version**
3.4.1

**Platform**
Linux

**Printer**
Anet A8

**Steps to Reproduce**
Printing any bigger project with multiple small holes and combining set to all. 

**Actual Results**
The walls of some of the small holes suffer from (under?) extrusion. Causing the walls to get half printed. 
 
**Expected results**
Walls printed normal.

**Additional Information**
- I'm using a 0.6mm nozzle.
- Combining set to aff fixes the problem.
- I'm not the only one, see: https://www.reddit.com/r/3Dprinting/comments/7if8yj/under_extrusion_after_travel_need_your_help/

amazingoomoo, 2018-09-06:
I too am having this exact problem and also have posted it on Reddit, in fact I was surprised when I realised that your Reddit link was not my post! This appears to be a Cura issue and it is ruining all my prints whenever there is a retraction and travel. Everything else prints fine. I am desperate for this to be fixed.

Ghostkeeper, 2018-11-06:
Does it help to set the "Max Comb Distance With No Retract" setting to some positive number? Like, 30mm?

closed 2020-06-20 by no-response[bot]

RainyCobra77982, 2023-09-03:
> I too am having this exact problem and also have posted it on Reddit, in fact I was surprised when I realised that your Reddit link was not my post! This appears to be a Cura issue and it is ruining [...]

Not involving this, but I'm running out of places to ask you, and this is probably my best bet at reaching you. Would you be willing to share the files for your wheatley project? I've been wanting to make one since around 2016 and finally have the means of doing it. Yours is the dream model, it checks all the boxes I've wanted in a real Wheatley, and I don't quite have the skills yet to model one myself of that quality. Hope you see this, thank you