Ultimaker/Cura #4293: One part after MERGE

mtmello2003, 2018-08-27:
After merge some parts, IMHO cura need to see a unique and fused one part but in fact, cura makes top solid infill and bottom solid infill with each parts merged, treating each one as individual.

This behavior causes also mistake with horizontal compensation, because cura see each part merged as individual and not as a unique fused to compensate in the right way.

fieldOfView, 2018-08-27:
Merge is not meant to do a boolean union. Merge intends to keep the meshes separate (but colocated), so you can apply different settings to each part of the whole.

If you want to do a boolean union, you have to do that in your modeling software of choice. Cura is not a modeler.

ChrisTerBeke, 2018-08-27:
Maybe we should call it 'group' instead of 'merge'...

fieldOfView, 2018-08-27:
There's already a "group". Grouping and merging are fairly similar in Cura. The difference is in positioning. Grouping keeps the position of each mesh as is. Merging puts the original origins of each mesh together before grouping.

ChrisTerBeke, 2018-08-27:
Mm. 'combine' maybe?

fieldOfView, 2018-08-27:
If the terminology is going to be changed (though it has been the same since legacy Cura so there are people used to it), I would try making it more descriptive. Eg "Group in place" and "Group around origin", though the latter may be confusing the origin of the buildplate vs the origin of the meshes.

Ghostkeeper, 2018-10-26:
> (though it has been the same since legacy Cura so there are people used to it)

This is to me a reason to keep calling it something with "merge". Though it should be mentioned that the original terminology in 15.04 and before was called "dual extrusion merge", which was referring to the intended use case for the feature.

I'm not able to reproduce the issue that you're describing about bridging. It doesn't do anything special with bridging for me. That would be a different issue anyway.

closed 2019-01-07 by nallath