Ultimaker/Cura #4294: Simple circular skirt taking 259hrs and off build plate

sproutdesign, 2018-08-28:
**Application Version**
<Mac 3.4.1>

**Platform**
<Mac OS 10.13.6>

**Printer**
<Ultimaker 3 Extended. Project file IP not shareable>

**Steps to Reproduce**
<changing infill %>

**Actual Results**
<This file would break the printer so we are not printing this>

**Expected results**
<No skirt expected, especially one taking 99% of the build time>

**Additional Information**
< [image]>

sproutdesign, 2018-08-28:
Just in case this demonstrates that the travels off the build plate are arbitary:
[image]

diegopradogesto, 2018-08-31:
Duplicate of https://github.com/Ultimaker/Cura/issues/4009

This is fixed in the next release! Thanks for reporting.

closed 2018-08-31 by diegopradogesto