Ultimaker/Cura #4297: APPCRASH Cura Crash 3.4.1 in my computer

gsohler, 2018-08-29:
I used to use Cura 15.4 in my computer but today i learned, that there are new versions available.

When I installed Cura 3.4.1 it crashes right away.

What can be the issue ?

Problemsignatur:
  Problemereignisname:	APPCRASH
  Anwendungsname:	Cura.exe
  Anwendungsversion:	0.0.0.0
  Anwendungszeitstempel:	5b113154
  Fehlermodulname:	atio6axx.dll
  Fehlermodulversion:	6.14.10.9606
  Fehlermodulzeitstempel:	4b7374ac
  Ausnahmecode:	c0000005
  Ausnahmeoffset:	0000000000d1cd83
  Betriebsystemversion:	6.1.7601.2.1.0.768.3
  Gebietsschema-ID:	1031
  Zusatzinformation 1:	bad0
  Zusatzinformation 2:	bad0f9149dde5085feb180048127c860
  Zusatzinformation 3:	26ac
  Zusatzinformation 4:	26ac2e0fca23bc8bba3838aef8940354

Lesen Sie unsere Datenschutzbestimmungen online:
  http://go.microsoft.com/fwlink/?linkid=104288&clcid=0x0407

Wenn die Onlinedatenschutzbestimmungen nicht verfügbar sind, lesen Sie unsere Datenschutzbestimmungen offline:
  C:\Windows\system32\de-DE\erofflps.txt

lf-, 2018-08-31:
Update or reinstall your graphics drivers. You deleted the template which told you to use dxdiag to provide more details on that, so that is all I can say with the limited details provided.

See "Fehlermodulname: atio6axx.dll". That is an AMD/ATI graphics driver.

gsohler, 2018-08-31:
Hi If and Diego.

Thank you for your valuable input. 
Just before i tried to create a new issue, but there was no hint about dxdiag in the template.
Still I can provide DxDiag.txt
[DxDiag.txt](https://github.com/Ultimaker/Cura/files/2339391/DxDiag.txt)

I have successfully reinstalled the Graphic driver from AMD for my 760G Card
Now i use Version 18.10 installed from amd-chipset-drivers_18.10_0813.exe

However, Cura still does not run: 
the appcrash is gone. it tells now: 

Could not Probe for OpenGL. OpenGL2.0 or higher required. Please check your video card driver.
Googling to download OpenGL 2.0 driver only shows Softonic spam and other links to amd itself, where i was before.

What are viable options now ?

gsohler, 2018-09-03:
problem is resolved. have a better graphics card now.
Thank you for your hints!

closed 2018-09-03 by gsohler