Ultimaker/Cura #4302: [3.4.1] Crash after Splashscreen

FranzSw, 2018-08-29:
**Application Version**
3.4.1

**Platform**
Windows 10 64bit (fresh installation)

**Printer**
None (Setup)

**Steps to Reproduce**
- Install Cura
- Start Cura once, it should open
- Close it with or without creating a profile for a printer
- Start Cura again

**Actual Results**
After the splashscreen, the main windows is visible for a very short time (<1s), then cura terminates

**Expected results**
No crash

**Additional Information**
[stderr.log](https://github.com/Ultimaker/Cura/files/2332010/stderr.log)

Critical part out of stderr:
```
2018-08-29 13:57:04,044 - DEBUG - [MainThread] UM.Qt.QtRenderer._initialize [183]: Support for Vertex Array Objects: True
2018-08-29 13:57:04,046 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [65]: An uncaught error has occurred!
2018-08-29 13:57:04,047 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]: Traceback (most recent call last):
2018-08-29 13:57:04,049 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]:   File "X:\3.4\build\inst\lib\python3.5\site-packages\UM\Qt\Bindings\MainWindow.py", line 215, in _render
2018-08-29 13:57:04,051 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]:   File "X:\3.4\build\inst\lib\python3.5\site-packages\UM\Qt\QtRenderer.py", line 120, in beginRendering
2018-08-29 13:57:04,052 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]:   File "X:\3.4\build\inst\lib\python3.5\site-packages\UM\Qt\QtRenderer.py", line 185, in _initialize
2018-08-29 13:57:04,054 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]:   File "X:\3.4\build\inst\lib\python3.5\site-packages\UM\View\GL\OpenGL.py", line 47, in __init__
2018-08-29 13:57:04,055 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]: AttributeError: 'NoneType' object has no attribute 'versionFunctions'
```

Information about graphics card:
[image]

MrKekson, 2018-08-29:
Happening all the time with me. Some settings got corrupted, and then it wont start ever. I version controll my settings.

FranzSw, 2018-08-30:
Hm, using version control is a good idea.

But still: Cura should not crash _that_ way. It should atleast try to catch the error and maybe give the user the option "Yes, I wanna reset my settings" to make it accessible for everyone. 

I have no trouble looking into the AppData, but some other users I know will 😄

ianpaschal, 2018-09-03:
@MrKekson This appears to be an issue with rendering and doesn't seem to have anything to do user settings/preferences.

MrKekson, 2018-09-04:
I use git for version controlling as i sad. And if it's keep crashing, and i roll back to the previous stable commit, it's fine again. Might be because of rendering, since it's happening after i see the windows, and it's loading the interface. That i why i commented, because the behavior is the same, crashing after the main win is visible. 
But resetting configs to a valid state surely solve it for me. 
Cura also constantly crashing if i run some games but not with all, and i found no pattern yet, HOTS is a 100% crash tho. Just get into a game, press slice and bamm, it's drops dead.

Ghostkeeper, 2018-12-13:
We switched back to Qt 5.9 a bit later in order to maximise support on MacOS. Let's hope it was fixed by upgrading to Qt 5.9, because we couldn't reproduce this problem on any of the testing computers in our office.

closed 2018-12-13 by Ghostkeeper

FranzSw, 2018-12-13:
I hope so, too.

I'm sorry, that I can't provide more information, but I'm away from home since 3 months and one more month, so I don't have access to my PC :/
@LipuFei