Ultimaker/Cura #4303: randomize item placement to even print bed wear

ChrisHeerschap, 2018-08-29:
Feature request

**Application Version**
Current

**Platform**
MacOS

**Printer**
Prusa i3 Mk3

**Steps to Reproduce**
Open a single model on an empty build plate. The model loads on the center of the build plate.

**Actual Results**
Models are located on the center of the build plate on load.

**Expected results**
Add an option to enable random start location of the model such that it is in the build  volume but in a random X-Y location

**Additional Information**
I recently realized that when I print single items, they're always located in the center of the build plate, which over time results in wear to that area of the build plate. To accommodate  this, I've started manually moving items to different places on the build plate, and I thought it would be handy to have the option to have Cura do this automatically. So, if the option is selected, any new item that's opened would be located somewhere in the build volume, just not at the center of the build plate. I think if multiple models were opened, having this feature disabled automatically would make sense, as you would otherwise be wasting plenty of time with moves between parts that are potentially spaced farther apart.

diegopradogesto, 2018-08-31:
I would say that this new feature won't be widely used since the user can always move the objects around. Maybe I'm wrong, so let this thread open as a feature request and wait for some more reactions.

ChrisHeerschap, 2018-08-31:
Yep, definitely. Not going to be a make or break feature, I move stuff around manually when I add it now anyway, but it'd be cool if it was easy enough to add and didn't get in the way of anything more important.

Ghostkeeper, 2018-10-30:
I'd argue against this. The centre of the build plate is often the best levelled location, so it's the most reliable until wear kicks in. With glass, aluminium or glue that's not an issue. With PEI sheets or something similar it is an issue but that might also just be an indicator that you should replace it.

fieldOfView, 2018-10-30:
It could be something a plugin would implement.

ReyvanZA, 2021-08-18:
Hi, well after printing a series of small PETG prints I have delaminated a very small section of my bed (dead center). I try to remember but occasionally start a print without moving the model. I have a new build plate replacement, but the existing one is almost perfect except for the small blemish.
Maybe not a random position, but an offset even?

ChrisHeerschap, 2023-06-30:
I've long since switched to Prusa Slicer so doesn't matter to me. 😆

GregValiant, 2024-11-24:
I'll go ahead and close this then.

closed 2024-11-24 by GregValiant