Ultimaker/Cura #4304: Reloading right side menu menu

MrKekson, 2018-08-29:
The right hand menu keep reloading. 
If i click the printer tab and back, it will happen. And it's reloading from the bottom, in a slow manner.

Can you please keep it in memory? Most PC got a ton of it, and i can't imagine how even hundreds of stringgs/nums/dropdowns can be this slow. Even in a browser it's well manageable. 

So please consider either keep it loaded, or speed up the loading/rendering, and start it from the top, where the most important settings are. It's just not gret to wait 10-20 sec, because i want to add 5 degree head temperature.

fieldOfView, 2018-08-29:
Note that this effect is a LOT less noticeable if you don't make all the settings visible. I guarantee you that you don't use them all. It is much more efficient to only keep those settings you use often visible, and use the search for the rest.

MrKekson, 2018-08-29:
The thing is. I do. I use multiple printer, and print a lot of different
things, with widely different settings, and material. So i don't use all of
them all the time, but i do regularly need to change most of them. That is
why i have it display all the settings. I might try to speed it up, but i'm
not that experienced with python.

diegopradogesto, 2018-08-30:
It was also reported in Uranium (https://github.com/Ultimaker/Uranium/issues/379) and before. We are looking into it. 

Internal ticket CURA-5652

Resonanz, 2018-09-15:
I can confirm that reloading the RHS is VERY slow (at least 5-10 seconds). Some of it appears with most text missing then it is slowly filled in.

Ghostkeeper, 2018-10-15:
This is fixed now. CURA-5652 was about the performance, but it also doesn't reload the setting item elements when switching tabs any more now.

MrKekson, 2018-10-15:
Great changes! Thanks!

closed 2018-10-15 by Ghostkeeper