Ultimaker/Cura #4305: Temp changes in gcode

Peggsy, 2018-08-30:
**Application Version**

Cura 3.4.1
**Platform**

Windows 7
**Printer**

Ender 3
**Steps to Reproduce**

Generating gcode for anything over 98 layers
**Actual Results**

gcode M104 inserted throughout the file. I set M104 S212 in Cura. at layer 98 this changes to 226. Then the temp is reduced by 4 degrees every 100 layers from there on.
**Expected results**

**Additional Information**

gcode file attached.
[SE-14R_-_Barrel.zip](https://github.com/Ultimaker/Cura/files/2335158/SE-14R_-_Barrel.zip)

alekseisasin, 2018-09-04:
@Peggsy , do you use any post-processing script?

Peggsy, 2018-09-04:
@alekseisasin Yep I found it.... well the problem was found for me, and did I feel dumb!!! It was a post-processing script I had left live after printing a heat tower.... sorry I should have closed the issue.
Thanks alekseisasin.

closed 2018-09-04 by Peggsy