Ultimaker/Cura #4308: Unnecessary retractions with Combing mode "Off"

dmitrygorbenko, 2018-08-30:
Hi,

When I set Combing mode to "Off" the Cura starts to use too many retractions while printing infills and top/bottom layers. And in some places this does not make any sense.

As an example I've took a screenshot of the bottom lines on layer 3 (0.7 mm) when the retractions happen with each bottom line (and these lines are not "external" ones). Take a look:

[image]

And this is the same place shown by http://gcode.ws :
[image]

And if we take a look at infill there are tons of retractions:

[image]

All of them don't make any sense there because the head does not move outside the model, it prints infills.

I understand what I can just disable retractions for this model :-)

But what if we improve retractions ?

Cura 3.4.1
Planform: Linux, Ubuntu
[top.3mf.gz](https://github.com/Ultimaker/Cura/files/2336697/top.3mf.gz)

Thanks,
Dima

FranzSw, 2018-08-31:
Just choose Combing: "No Skin"?

dmitrygorbenko, 2018-08-31:
yeah, but these retractions don't make any sense there, do they ?

smartavionics, 2018-08-31:
Increasing the value of Retraction Minimum Travel will stop those retractions from occurring.

dmitrygorbenko, 2018-09-04:
Sure, there are many ways to prevent these retracts from happening, but the main issue is what some core slicing algo is producing them.
They should never happen at all at these points, no matter what retraction-related settings do I use.

agibson2, 2018-09-19:
The info I wrote was incorrect.  I forgot that when I noticed it, I was not printing infill.  I was printing the first 3 layers of a print.  Infill generally doesn't print solid layers.  I had left Coasting set to off after the Gyroid infill testing.

Ghostkeeper, 2018-11-19:
If combing is disabled, CuraEngine doesn't check whether the travel move it makes ever hits any walls. It could, and I agree that would be an improvement.

We've decided though that this has no priority for us yet since it's easy enough to just enable combing. That will usually give a better result anyway. We'll put it on the list for "one day maybe".

artz, 2019-10-30:
Is anyone else missing the `Retraction Minimum Travel` setting in Cura 4.3.0 (Mac OSX version)?  

[image]

GregValiant, 2026-09-06:
I'm cleaning up some of these old reports.  I'll close this.  Currently 5.13.0 is the latest Cura version.

closed 2026-09-06 by GregValiant