Ultimaker/Cura #4310: [3.4.1/commit ae1a3606] Support Blocker appears not to work

lf-, 2018-08-31:
**Application Version**

3.4.1, also git version ae1a3606ba835c194eebc61058a4d0c34dc81e4e

**Platform**

Arch Linux x86_64

**Printer**

Attached, file is a 3mf with changed extension: [supportbug.3mf.zip](https://github.com/Ultimaker/Cura/files/2338495/supportbug.3mf.zip). Affected STL is also attached, again renamed to .zip but is actually an STL: 
[knob2.stl.zip](https://github.com/Ultimaker/Cura/files/2338496/knob2.stl.zip).

**Steps to Reproduce**

Attempt to use support blocker to block supports on center section of knob model attached.

Here's a video: https://youtu.be/WrENQq91QE8

**Actual Results**

Supports are not blocked.

**Expected results**

Supports should be blocked.

**Additional Information**

fieldOfView, 2018-08-31:
In other words, the support blocker cubes are not simply subtracted from the support. What they do is "mark" parts of your mesh that should not get supported.

lf-, 2018-08-31:
Oh I see. Thanks for doing that! I wonder though if I uncovered a potential UI improvement though, with a section view or something for this purpose.

lf-, 2018-08-31:
Also shouldn't it be suppressing supports on the bottom of the dial as well? That's what prevented me figuring out how the feature worked: there were supports even on the flat parts around the center piece which definitely intersected the blocker.

fieldOfView, 2018-08-31:
That was not support for the centerpiece. The part of the centerpiece that intersected with the blocker did not need support.