Ultimaker/Cura #4314: [master] weird fat skin lines

smartavionics, 2018-08-31:
**Application Version**

master

**Platform**

Linux

**Printer**

Kossel XL

**Steps to Reproduce**

Open cura, select printer, select profile, load model, slice.

**Actual Results**

Every other skin layer has a weird region where the lines are too fat and some lines are missing.

**Expected results**

Normal skin

**Additional Information**

The region that is bad is at the start of the skin.

[image]

smartavionics, 2018-08-31:
Here's the project, it's not a standard printer, hope you can load it...

[KXL_glasses-leg.3mf-renamed-to.zip](https://github.com/Ultimaker/Cura/files/2341532/KXL_glasses-leg.3mf-renamed-to.zip)

diegopradogesto, 2018-09-04:
Don't worry, it's very easy to reproduce even with a Custom FFF Printer.

diegopradogesto, 2018-09-04:
Devs.: CURA-5690

LipuFei, 2018-09-14:
@smartavionics Also found that in your `kossel_xl_extruder_0.def.json`, `material_diameter`'s `default_value` should actually be `1.75` instead of `"1.75"`.

```
    "default_value": "1.75",   Change it to --> 1.75 without double quotes
```

smartavionics, 2018-09-14:
OK, thanks a lot for spotting that.

smartavionics, 2018-09-14:
Shouldn't all the settings be in the project?

If you can't reproduce it, let's assume it was a glitch and close this issue, if I find it happening again, I will reopen.

diegopradogesto, 2018-09-14:
Try with this project file and see layer 13 for example.

[CFFFP_example.zip](https://github.com/Ultimaker/Cura/files/2383142/CFFFP_example.zip) (Rename to 3mf)

smartavionics, 2018-09-15:
With my cura it slices but fails to display the layer view and I get these messages:

````
2018-09-15 11:28:40,932 - ERROR - [Thread-3] UM.Logger.logException [84]: Exception: Job <CuraEngineBackend.ProcessSlicedLayersJob.ProcessSlicedLayersJob object at 0x15aca5a8> caused an exception
2018-09-15 11:28:40,932 - ERROR - [Thread-3] UM.Logger.logException [88]: Traceback (most recent call last):
2018-09-15 11:28:40,933 - ERROR - [Thread-3] UM.Logger.logException [88]:   File "/opt/cura/lib/python3/dist-packages/UM/JobQueue.py", line 121, in run
2018-09-15 11:28:40,933 - ERROR - [Thread-3] UM.Logger.logException [88]:     job.run()
2018-09-15 11:28:40,934 - ERROR - [Thread-3] UM.Logger.logException [88]:   File "/opt/cura/lib/cura/plugins/CuraEngineBackend/ProcessSlicedLayersJob.py", line 192, in run
2018-09-15 11:28:40,934 - ERROR - [Thread-3] UM.Logger.logException [88]:     material_color_map[position, :] = color
2018-09-15 11:28:40,934 - ERROR - [Thread-3] UM.Logger.logException [88]: IndexError: index 1 is out of bounds for axis 0 with size 1
````

smartavionics, 2018-09-15:
I updated my cura+uranium to the current masters and the above failure goes away. Furthermore, I am not seeing the fat skin lines.

I vote that we close this issue for now and if I see them appear again, will reopen it. Thanks for looking at it anyway!

LipuFei, 2018-09-19:
@smartavionics It's caused by this commit: https://github.com/Ultimaker/CuraEngine/commit/06dc32c224ed28b60bae69df5f031e82442fbe0c

smartavionics, 2018-09-19:
Hi @LipuFei , well the bug is in the 3.5 beta and is biting people!

LipuFei, 2018-09-19:
We are on it.

smartavionics, 2020-07-14:
No longer a problem for me.

closed 2020-07-14 by smartavionics