Ultimaker/Cura #4315: Fill in unexpected areas

nauzadt, 2018-09-01:
**Application Version**
3.4.1

**Platform**
Win10

**Printer**
Cocoon HE170703 / Wanhao Duplicator i3 mini

**Steps to Reproduce**
Load model.
View model in solid view and Layer View

**Actual Results**
[image]

**Expected results**

[image]

**Additional Information**
Seems like Cura wants to fill in the gap in the lettering

ianpaschal, 2018-09-03:
Please save your project as a .curaproject, zip it, and attach it. Very hard to guess what is going wrong looking at a picture.

nauzadt, 2018-09-04:
Hi Ian, I have a .stl file that has lettering in it. For some reason, this version of Cura wants to fill the inside space within the letter. .

The "actual results" is a screenshot of a layer view, while the "Expected results is a screenshot of the .stl file.

ianpaschal, 2018-09-04:
Yeah, I see that. ;)

Please save your project as a .curaproject, zip it, and attach it. I see something is going wrong but can't tell what if I can't look closely at the model, configuration, etc.

nauzadt, 2018-09-08:
@ianpaschal Reloaded the file. Its generating the right STL now. Thanks

closed 2018-09-08 by ianpaschal