Ultimaker/Cura #4317: Narrow horizontal slot shell processing bug

Dabihu, 2018-09-02:
Cura 3.4.1
windows 10 x64
Marlin

Steps to Reproduce:
If a horizontal slot is in the model, the Shell/ Top/Bottom Thickness could cause empty/infilled layers.
Condition: The slot is less than Top/Bottom Thickness.

The attached sample file has a 0.8 mm height slot:
[Sample.zip](https://github.com/Ultimaker/Cura/files/2343450/Sample.zip)

Layer height: 0.2,  Top/Bottom Thickness: 0.8 mm (slot = thickness: good)
[image]
Layer height: 0.2,  Top/Bottom Thickness: 1.6 mm (slot < thickness: bad)
[image]

diegopradogesto, 2018-09-04:
Easily reproducible, sounds like a bug. Thanks for sharing!

diegopradogesto, 2018-09-04:
Devs.: CURA-5689

rburema, 2018-09-13:
Hi @Dabihu,

It turns out that this currently is expected _(default)_ behaviour, but you can change this by unchecking 'Ignore Small Z Gaps' in the settings. Given your model, this'll increase both slicing and printing time only by a very manageable amount. On more complex models though, it can change (especially) slicing time by more than 10%.

The team will discuss whether to flip the default behaviour shortly (since you brought it to our attention again), but otherwise I'll close the issue.

If this persists despite unchecking 'Ignore Small Z Gaps', feel free to reopen.

cheers, Remco.

closed 2018-09-13 by rburema

Dabihu, 2018-09-18:
It's working fine. Thank you!