Ultimaker/Cura #4318: Circle holes dimension problem

kyjak, 2018-09-03:
**Application Version**
3.4.1

**Platform**
Windows 10

**Printer**
Prusa mk3

**Steps to Reproduce**
Print model with different sizes of holes, for example 2, 4, 8, 12 mm.
The smaller the hole, the more the diameter is deviated from the design (printes hole is smaller). I calibrated my printer, all dimensions inclusing square holes are OK.
The problem probably lies in wiping the filament to the center of the hole while moving the head in circle. The horizontal offset in cura setting does not help, since it destroys the size of rectangular holes.
I suggest to add circular offset, based on diameter as a setting.

**Actual Results**
Circular holes are always smaller.

**Expected results**
Exact hole diameter as designed.

**Additional Information**
I would give this a top priority :) as good prints are what we all are looking for.

ianpaschal, 2018-09-03:
Are you sure this is not caused by STL resolution?

smartavionics, 2018-09-03:
This has been a problem with FDM printing for years, one solution that actually works quite well is to reduce the number of vertices in the hole outline depending on the hole diameter. The smaller the hole, the fewer the vertices.

Here's an openscad routine that implements this idea:
````
// Copyright 2011 Nophead (of RepRap fame)
// This file is licensed under the terms of Creative Commons Attribution 3.0 Unported.

// Using this holes should come out approximately right when printed
module polyhole(h, d) {
    n = max(round(2 * d),3);
    rotate([0,0,180])
        cylinder(h = h, r = (d / 2) / cos (180 / n), $fn = n);
}

module test_polyhole(){
difference() {
        cube(size = [100,27,3]);
    union() {
        for(i = [1:10]) {
            translate([(i * i + i)/2 + 3 * i , 8,-1])
                polyhole(h = 5, d = i);
                
            assign(d = i + 0.5)
                translate([(d * d + d)/2 + 3 * d, 19,-1])
                    polyhole(h = 5, d = d);
        }
    }
}
}
````

[image]

DDDirk, 2018-09-03:
I can't read openScad, but it seems you only use a multiplier, resulting in too few segments for small holes. For that reason I also use an adder, such that:
numberOfSegments = multiplier * radius + adder
The multiplier mostly affects the bigger radii and the adder the smaller radii. Effectively, the adder sets a minimum number of segments. This may result in too many segments for very small holes, so it shouldn't be set too high. Or a condition might be added checking segment length, and setting that to a maximum value if it is exceeded.

smartavionics, 2018-09-03:
It's not my code, just something that has been around for years which attempts to solve this problem. I can remember using it in the past with reasonable results.

kyjak, 2018-09-03:
The closest thing that might is the cura stretch plugin. I thought to include it in the slicer process, test the results, create controlling variables and actually enable this setting. As you say,long standing problem, its time to solve it the proper way, not workarounds...

ianpaschal, 2018-09-03:
We will discuss it as a team but what many hackers and hobbyists don't know is that in the industrial world, adapting your CAD to the idiosyncrasies of your CAM is a universal truth. For example, when extruding hot steel through a die (like similar to the machines that smoosh Playdoh out of a star-shaped hole), the die is not actually the intended shape; it's something similar but adjusted to compensate for the nature of working with gooey red hot steel. FDM is not so different I don't think it's unreasonable at all to suggest that you might have to adapt the design slightly to the manufacturing process. In any case: we'll discuss it.

ChrisTerBeke, 2018-09-03:
Even injection molding requires design optimizations depending on the molding technique and material. It's really quite common in any manufacturing process. It's just much more present in FFF printing though as the process and materials are designed for flexibility and speed rather than part fit. I think both the CAM design as the shape estimation by the slicing engine (by giving it more information about the printing process, currently not possible) are needed to improve it.

ianpaschal, 2018-09-04:
We have talked about this as a team.

Ultimaker is in the midst of a long, ongoing hardware-software investigation into "dimensional accuracy" solving these sorts of problems. That's really all we can do right now, and so in light of this not really being an issue which can "close" at some point (there's always going to be room for improvement), I'm going to have to close this.

As suggested by the team member working in the dimensional accuracy investigation, for the time being you can mitigate the problem by printing outer walls before inner walls (that is to say: the surface touching air vs. the surface touching infill). This will help ensure that the outer dimensions of your print match what you expect more closely.

But in truth, part of the appeal of rapid manufacturing/prototyping is that identifying and compensating for these idiosyncrasies is far easier and cheaper than doing so with an injection mould which costs thousands of euros/dollars to tool.

Just print, check, adjust, print again. I'd say "that's the hacker/maker way" but it's actually pretty much the way all engineering and design in the world is done.

closed 2018-09-04 by ianpaschal

kyjak, 2018-09-04:
I successfully used today the stretch plugin to offset curved paths and get 0.05mm accuracy of curved holes.
The point is, this can be moved from external plugin to the main setting menu, so it's more handy to use and actually can be saved as a profile setting. This should not require much work.

Ghostkeeper, 2018-11-05:
The stretch script is not a stable solution. It produces some weird lines sometimes. More importantly, it's not the type of solution we'd like to use: It's safer to expand the polygons after slicing, before generating paths to fill them, than expanding the lines after the g-code is generated. Otherwise your small gap fillings and such will be off.