Ultimaker/Cura #4322: Import and Export

Romzik85, 2018-09-03:
After exporting the profile, the program closes itself. After importing the profile, the profile is displayed, after which the program closes itself, upon reopening the imported profile is absent.

**Application Version**
3.4.1

**Platform**
Windows 10, 1809 Build 17751.1

**Printer**
Custom (D-Bot, Lerdge-S mainboard)

diegopradogesto, 2018-09-04:
Can you please attach the log files so we can see what happened? You can find the **cura.log** file in the following directory: C:\Users\<username>\AppData\Roaming\cura\3.4.

You can also attach the whole 3.4 folder compressed as a zip, so we can reproduce your exactly environment.

Romzik85, 2018-09-04:
[cura folder](http://artiromiur.ru/files/cura.zip)
I can not find something, can I attach a file here?

diegopradogesto, 2018-09-04:
I've just loaded your configuration and all seems to be fine to me. I see the profile and I can correctly slice a model. Besides that, I don't see any errors or crashes in the logs, so I don't know what happened. Can you easily reproduce the error? Could you please list the steps, one by one?

Romzik85, 2018-09-04:
it's easier to show on [video](https://www.youtube.com/watch?v=u6IzEsmfb30)

Romzik85, 2018-09-04:
and I tried to import this file
[мой.zip](https://github.com/Ultimaker/Cura/files/2350089/default.zip)

diegopradogesto, 2018-09-11:
I redo the steps that you show in the video, with your configuration data and I don't see any crashes. Everything works correctly for me. It seems to be dependent of your pc/laptop.

Let's see if somebody else can reproduce the issue.

closed 2018-09-11 by Romzik85