Ultimaker/Cura #4323: Spiralize Feature request

gsohler, 2018-09-03:
Hi Group,

I like the Spiralize Feature very much. It saves a lot of printing time and in most cases
it only matters what you see, so only print surfaces.

I did some tests with the spiralize feature and i found out, that the walls will drop, if the surface angle bends more that e.g. 40 degrees.
Is it possible to specify a number of walls for spiralze, which is different from one ?
For each Layer cura could extrude all the layers from inner to outer. I hope that using this technique the print would be more safe agains walls dropping(which in fact does not recover very well)
I could not test it of course, but in't it worth a trial ?
P.s. in a second step - maybe its possible to adapt the number of walls dependant on the wall angle ?

**Application Version**

**Platform**

**Printer**

**Steps to Reproduce**

**Actual Results**

**Expected results**

**Additional Information**

smartavionics, 2018-09-03:
Hi @gsohler , that's an idea I have never considered before. Not sure it's feasible. However, wouldn't a model that has 2 or more walls but 0% infill achieve the same goal?

gsohler, 2018-09-03:
Hi Smartavionics,
Thank you for your comment.

My understanding so far was, that the "spiralize" option uses an algorithm separate from the normal
extrusion procedure, which strictly advices the printhead to rotate CCW e.g whereas for the normal procedure this is not a restriction as it only tries to minimize moves between extrusions.

In order to elaborate this topic any further, I need to back-ask you: what is the actual difference between "spiralize" option and 1 normal extrusion procedure with 1 wall and 0% infill ?

Günther

gsohler, 2018-09-03:
Hi smartavionics

Thank you for your explainations.
So i understand, that the main difference between spiralize and ordinal is that in spiralize, the z coorindate will smootly rise whereas in ordinarr mode, the z height will jump after every layer by one layer height.

In spiralize you could still  raise z height smootly even when using multiple walls

This is an example of one wall(| is wall)
| (1.0)      center (1.5)|
|(2.0)                  (2.5)|
|(3.0)                  (3.5)|

And this is for 2 walls. This one extrudes the inner wall first

|(1.5)    |(1.0)              (1.25)|      (1.75)|
|(2.5)    |(2.0)              (2.25)|      (2.75)|
|(3.5)    |(3.0)              (3.25)|      (3.75)|

As you can see, even with multiple walls, layers on top of each other, will always have an exact layer distance of 1.0. Only issue here is when moving from outer to inner once every layer at a local spot,
extrusion is "ironed" , but i think this does not harm as the extruder is hot and will melt again

What do you think about that ?

smartavionics, 2018-09-03:
Sorry, I can't work out what you are showing in the diagrams. To make things easy, let's assume the object is a square and so each wall in each layer only has 4 points. Using the notation LWP = Z (layer,  wall, point) = height, please show the order for two walls like this example for 1 wall:

000 = 0
001 = 0.25
002 = 0.5
003 = 0.75
100 = 1.0
101 = 1.25
102 = 1.5
103 = 1.75
200 = 2.0
...

Thanks.

smartavionics, 2018-09-03:
OK, I understand now your idea now. My gut feeling is that it doesn't work but I'll think about it some more and get back to you. Cheers!

gsohler, 2018-09-03:
If you spiralize with one wall, the plastic extrusion  falls down if it  misses to hit

*  the layer below

If you spiralize with two walls. the plastic extrusion falls down if it misses to hit

* the layer below
AND
* the layer to the side
AND
* the layer below to the side

So theoretically, chances are 4x less that the plastic extrusion walls for an angled wall.
This is why i belive its an adavantage

diegopradogesto, 2018-09-04:
I like the idea to be able to have a spiralize feature with several walls. The procedure seems OK to me, except that it will create an artifact when in the new layer the nozzle moves from the outer wall to the inner wall again. But maybe it doesn't affect the quality too much.

gsohler, 2018-09-04:
Yes, i mentioned that artefact before. However I doubt you will see it as the the nozzle will extrude an outer wire at the each colission place, finally

smartavionics, 2018-09-04:
I struggle to see how this can work because having printed the inner wall, when it starts the outer wall the nozzle will be 1/2 layer height too high (i.e. the gap between the nozzle and the previous outer layer will be 1.5 times the layer height.

gsohler, 2018-09-04:
smartavionics, we are talking about this secion in the table:

103 = 1.375
110 = 1.5

Here it is moving from the inner wall to the outer wall. New position is 1.5
The Layer just below is:

010 = 0.5

It has an exact distance of 1 layer height

Guenther

smartavionics, 2018-09-04:
Duh!

smartavionics, 2018-09-04:
OK, here's a new problem to consider... The inner walls are created by shrinking the outer wall. Think what happens when a shape like an hourglass is shrunk. Where the shape narrows in the middle, the walls can disappear when the shape is shrunk so the inner wall is actually two separate walls. So now you have a single outer wall enclosing two separate inner walls. What do you do there?

gsohler, 2018-09-04:
Hi Smartavionics,

you are right, thats defintely a problem to consider.
1st of all: You cannot Spiralize ervery object. Basically this only makes sense for objects in which  for each layer there only exist exactly one cross-section. This mostly applies for Vases and extrusion objects.
Of course it can happen, that you get two objects when shrinking an hourglass, which was one object before, but its unlikely because you only shrink by 1 or 2 nozzle widths.

But of course it can happen. To mitigate that i suggest,  to select the object with the greatest perimeter in case shrink yields more than one result. This would result in onle layer partially only having one wall,
but still much better than the whole print only having one wall.

What do you think ?

smartavionics, 2018-09-04:
I think in that situation you have to choose the inner wall that contains the vertex that is nearest to the first point in the outer wall.

gsohler, 2018-09-04:
of course. you cannot accept moving far away in order to meet the desired inner wall. this would definitely show disruptions in the print.

DDDirk, 2018-09-05:
I have been thinking about this too, but it won't be as 'clean' as a spiralise, which doesn't have a z-seam. Here, some disruption is unavoidable when changing between the outer wall and the innermost wall (and vice versa).
But first, how is it done?
In the example of gsohler, for a square object:
000 = 0
001 = 0.125
002 = 0.25
003 = 0.375
010 = 0.5
The move between the last two is at an angle, because there is an offset between the inner and outer wall. And the same when it moves inward again:
013 = 0.875
100 = 1
This is also at an angle, in the other direction. These moves create a dent in the front. But also, with the second one, the road crosses the first angled road, thus:

file:///home/dirk/spiralise_helicise_1_1.png 

And then, after the inner wall is done, you get the same going outwards. This can be solved by first moving to the corner of that wall and only then to the corresponding corner of the other wall.

And maybe this will give better results than staying at the same layer height because you cross the previous roads at half layer height, so you don't have to stop extrusion completely. So in theory a G1 should be done with the extrusion value halved, but there is always a delay in that. But because of that a G0 with the standard layer-per-layer can create a blob, and the fact that there is some room for the ooze this might be prevented.

DDDirk, 2018-09-05:
Hmmm, I couldn't get the image inserted and I 'edit' also doesn't work. Weird. Hopefully the description is clear enough.

DDDirk, 2018-09-06:
Yes, after writing that I also realised that a (near) right angle is not a good idea because it affects the print speed, probably reducing it to the jerk speed of 20 mm/s. This would likely result in overpressure in the nozzle and possibly a blob. Which might solve the problem of the dent. :) The best angle at which to do this (so at which point the intermediate vertex should be placed) is to be determined experimentally and will likely depend on the (nominal) print speed (among other things).

gsohler, 2018-09-06:
I think, smartavonics started to code for this feature. I am wondering if he was able to collect 1st experiences

smartavionics, 2018-09-06:
Hi folks. Yes, I did start to code but have been busy on "real" work since then. Actually, I can forsee some problems implementing this due to how the existing cura code works so I can't really predict how long it will be before I have managed to do some experiments. Anyway, I won't forget about it, please be patient.

gsohler, 2018-09-19:
Some time gone bye . Would be interesting to know which challenges are ahead right now for progressing with the double spiralize ....

Ghostkeeper, 2018-11-16:
Time for experimentation, really. Spiralize doesn't typically have a high priority for us because we don't see a lot of people using it and the current implementation works mostly fine.