Ultimaker/Cura #4324: Cura 64 Bit ONLY Crashes on startup

morsyfadl, 2018-09-03:
**Application Version**
Any 64-Bit version. I tired 3.4.1, 3.3.1, 3.2, 3 and  2.7
**Platform**
Windows 8 64 Bit
 i3-2120
 Intel HD 2500
2GB DDR3 Ram

**Printer**
Custom Prusa (irrelevant)

**Steps to Reproduce**
Install any version that is 64 bit, and try to run it right after installing, even on a clean windows with the appdata/roaming files deleted

**Actual Results**
cura barely opens a white windows - doesnt really load- and give a "has stopped working" error
Meanwhile, Every single 32-bit version that i tried works flawlessly, from 15.06 to 2.3

**Expected results**
It should load normlly

**Additional Information**

alekseisasin, 2018-09-04:
@morsyfadl , probably you have a problem with the graphical card drivers. Please upload log file, maybe it will say more information. I am not sure about your virtual memory, might be also not enough.

alekseisasin, 2018-09-11:
I used to close the issue because I assume it is related to your computer hardware. Otherwise, we would have similar issues from Cura 2.7 version.

closed 2018-09-11 by alekseisasin