Ultimaker/Cura #4325: Creating .curaprofile files externally

BaconBad, 2018-09-04:
Hi,

I am looking for a way to create .curaprofile files externally and then import them in Cura profile manager.

I have managed to open up a basic .curaprofile file I exported from the profile manager and access the containers inside, but I struggle to understand the structure of these containers, and in turn, to modify them correctly. I have also looked into fdmprinter.def.json and fdmextruder.def.json files to locate all the variable names, their intended units, limits and defaults, however, jamming all the variables with their respective values in a .curaprofile container yields a corrupt profile.

Here are the things I do not understand:

- Why are there 8 containers in a .curaprofile file?
- Why do most of the containers contain only the same metadata and version information, and only two of them contain the actual printing settings?
- Why are the printing settings distributed across the containers the way they are?
- Do [fdmextruder.def.json](https://github.com/Ultimaker/Cura/blob/master/resources/definitions/fdmextruder.def.json) and [fdmprinter.def.json](https://github.com/Ultimaker/Cura/blob/master/resources/definitions/fdmprinter.def.json) define the same parameters which exist in .curaprofile files?

My apologies if this violates the issue submission guidelines, I do not have any better ideas on where to submit questions 

**Application Version**
Cura 3.4.1 x64

**Platform**
Windows 10 x64

fieldOfView, 2018-09-05:
A “Custom FDM Printer” has up to 8 extruders. Even if you have enabled only a single extruder at the time of saving a curaprofile file, Cura saves a container for each extruder. The “extra” extruders typically don’t have settings (but do have some metadata).

Some settings are set per extruder, but some settings are set globally. So there’s an additional container with global settings. That is the second container that has setting values.

The inst.cfg files have a very similar structure to the files in the “user” folder of the configuration folder. Those files contain the changed settings that have not yet been stored in a profile.

nallath, 2018-09-07:
fdmextruder.def.json defines settings that are only relevant to an extruder. For example; the offset of the extruder from the head (Which is a setting that only makes sense for the extruder)

nallath, 2018-09-10:
True, but doing that will probably create more issues than it solves. I already dread having to make an upgrade script for that...

Ghostkeeper, 2018-11-09:
To answer some of the other questions directly:

> Why are there 8 containers in a .curaprofile file?

There is one for each extruder, and one extra for the global settings that cannot be changed per extruder (such as bed temperature). For Custom FFF printer however, most extruders are hidden by default, so actually there are 8 extruders even though you appear to have only 1. This is a technical limitation in Cura.

> Why do most of the containers contain only the same metadata and version information, and only two of them contain the actual printing settings?

Because you only modified actual printing settings in one or two extruders and the global settings, probably. The profiles that are empty.

> Why are the printing settings distributed across the containers the way they are?

We need to keep track of which settings are modified for which extruders.

> Do fdmextruder.def.json and fdmprinter.def.json define the same parameters which exist in .curaprofile files?

Almost. They define the same settings, but they define more of it, because they also define the name, description, maximum value, etc. Modifying the setting value in a profile should be equal to modifying the `value` property of the settings in a `.def.json` file.

closed 2020-01-30 by nallath