Ultimaker/Cura #4327: Usability issues when tabbing between fields.

kirilledelman, 2018-09-04:
**Application Version**
3.4.0, 3.4.1

**Platform**
Mac OS

**Printer**
Custom FDM Printer

**Steps to Reproduce**
Tab between numeric text fields.

**Actual Results**
Tabbing into a new numeric text field places cursor to the end.

**Expected results**
Selecting text in field so I could type over it.

**Additional Information**
It's really annoying and doubles the amount of keystrokes needed to quickly edit values in the profile, printer settings, or X, Y, Z values in Scale & Position popups in the preview.

fieldOfView, 2018-09-04:
Is it working correctly in the (custom) sidebar for you?

kirilledelman, 2018-09-04:
@fieldOfView It does - it works well on the right side fields, but not in the transform windows, or printer settings.

fieldOfView, 2018-09-04:
Ok, that is good to know, AFAIK, the transform panels will have proper tabbing behavior in the next release. I'll have to check the Machine Settings window.

Ghostkeeper, 2018-11-05:
In 3.6 (right now at least) it selects the text when pressing tab in the move tool panel. Tested on Ubuntu 18.04.

closed 2018-11-05 by Ghostkeeper