Ultimaker/Cura #4328: No confirmation when asking to reset Cura to factory settings.

kirilledelman, 2018-09-04:
**Application Version**
3.4.0

**Platform**
Mac OS

**Printer**
Custom FDM Printer

**Steps to Reproduce**
At application start Cura displays a notification on the bottom that my printer profiles are corrupt (how that happened I don't know). It's a large message that covers half of the screen (lists multiple old profiles, some were deleted). The message can be dismissed by clicking X in upper right. The message also has a button "Reset Cura to defaults" or something like that. I clicked it accidentally instead of X and it wiped all my printer, carefully tuned profiles, and materials without confirmation. Hair pulling disaster event!

**Actual Results**
Cura doesn't display "Are you sure you want to delete all your custom printer profiles, print profiles, and materials?"

**Expected results**
Cura should confirm "Are you sure you want to delete all your custom printer profiles, print profiles, and materials?" before self destructing.

NicholasTracy, 2018-09-07:
I can confirm that this is an issue that needs to be addressed. I just experienced this with 3.4.1. The error in question popped up after duplicating the generic material profile and changing the default temp for the duplicated profile. When clicking reset Cura actually crashed and I had to restart Cura, when it restarted I had to start over from scratch on all my configuration settings and profiles. Very frustrating.

nallath, 2018-09-07:
This issue is in our sprint right now; CURA-5692

nallath, 2018-09-13:
See; https://github.com/Ultimaker/Uranium/commit/518664931ff5daf9aa13c906e4a879f74f90dd04

closed 2018-09-13 by nallath