Ultimaker/Cura #4329: Click to resume in the middle of print

kirilledelman, 2018-09-05:
**Application Version**
3.4.1

**Platform**
MacOS

**Printer**
Custom FDM / Marlin / PRUSA clone
[CFDMP_shutter_plain_RESUME.3mf.zip](https://github.com/Ultimaker/Cura/files/2354745/CFDMP_shutter_plain_RESUME.3mf.zip)

**Steps to Reproduce**
Print the file

**Actual Results**
Printer stops on its own at 2.5mm level with a message "Click to resume...".

**Expected results**
No pause.

**Additional Information**
Printer appears to wait for interaction for no apparent reason. While it sits there, unattended, a big glob of PLA slowly builds up around the nozzle.

diegopradogesto, 2018-09-06:
Do you have the "Pause at height" post-processing script enabled?

fieldOfView, 2018-09-06:
Probably a duplicate of #3994

kirilledelman, 2018-09-06:
I don't have that script installed or enabled. Mine always pauses at the beginning of print. Pausing in the middle of print happened once yesterday. My heated bed temp is set to 0.

kirilledelman, 2018-09-18:
Still happens, multiple times per print.

sabsteef, 2018-10-10:
Same here, got an ultimaker clone and pause mid print multiple times. im also on a mac os platform.
ang got Marlin firmware 1.19  and ultimaker 3.5 please look in to the mater.

Alef34, 2018-11-05:
I have the same problem, Cura 3.5.0, Win10, Marlin 1.1.8, RAMPS 1.4, Arduino Mega and RepRap printer.

fieldOfView, 2018-11-05:
It has been fixed in Cura 3.6, which is currently in open beta.

KnifeyMcShotfun, 2019-03-05:
I'm having this problem in v3.6.0.

Printer is Anet A8 via USB, Cura 3.6.0, Win 10, Marlin 1.1.9
(Printer runs with Mosfets and resoldered connections to make it less of a fire hazard, but I don't think that'll affect this?)

I'm a bit new to Github, do you need anything else from me?

Ghostkeeper, 2019-03-06:
Well, we need to be able to reproduce the problem in order to fix it, unless it is obvious to be seen from the source code.

So if the problem only occurs in an Anet A8, then we'd need an Anet A8 and I don't have one. Or you'd need to investigate yourself how we can fix the problem in the source code. We're basically relying on the community to fix issues with USB connectivity of specific printers.

KnifeyMcShotfun, 2019-03-11:
Hi Ghostkeeper,

Thanks for the response and apologies for the delay in getting back to you.

I've managed to work around the issue by generating the gcode in cura then performing the print itself via a different software package - unfortunately I'm not savvy enough to give any other suggestions for fixes.